citruz/haos-rockpi #1: Feature request: 4B without the +

mthqwork, 2023-01-15:
### Describe the issue you are experiencing

This build looks very cool. However it wont runs on 4B without the + as it's different.
Any chance that you can add support for the 4B? :)

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

Latest

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Put image to the SD
2. It won't boots, no picture on HDMI.

### Anything in the Supervisor logs that might be useful for us?

```txt
----------------------------------------
```

### Anything in the Host logs that might be useful for us?

```txt
----------------------------------------
```

### System information

Rock Pi 4B

citruz, 2023-01-29:
Sorry for the delayed response, for some reason I did not get a notification.

I am a bit confused that it does not work on the 4B since those two board are veeery similar and users have reported that this build even works on the 4SE and 4C+ which are more different from the 4B+. For example, Armbian's 4B build also works on the 4B+ except that the cpu frequencies are reported incorrectly because of the slightly newer CPU.

I can create a build with the 4B device tree, but I am honestly not sure if that will solve it. Can you give some more info about your setup? Eg what is the size of the SD card? How are you flashing it?

Do you have serial to USB adapter? That would be great for debugging boot issues.

mthqwork, 2023-01-29:
I have flashed it in many different ways including DD at Linux and imgUSB on Windows. Looks like the board tries to boot, but no output on HDMI and activity LED does not flashing. Armbian working fine :) IIRC it was a 64 GB Kingston Canvas, same used for the Armbian.
Yes, i do have, but im not sure how to wire it up.

citruz, 2023-01-29:
Just to make sure, you extracted the image before flashing?

mthqwork, 2023-01-29:
Sure!

citruz, 2023-01-29:
Ok just wanted to check 😅
Please see the radxa site for instructions on how to setup the serial interface: https://wiki.radxa.com/Rockpi4/dev/serial-console

In the meantime I am doing a build specifically with the 4B devicetree so that we can check whether that is the issue.

citruz, 2023-01-30:
Please try the following image: https://cloud.felixseele.de/index.php/s/9t4qb8keK9s8dK2
It's explicitly built for the 4B.

citruz, 2023-02-05:
Any news? There is also a new release which uses a more recent bootloader. Please try that image.

mthqwork, 2023-02-05:
I had no time to test it, but I'll do it today, just thought to check it. Do you have some links to the new one? Thanks

citruz, 2023-02-05:
It's not built specifically for the 4B, but should work too: https://github.com/citruz/haos-rockpi/releases/tag/9.5%2B20230204

mthqwork, 2023-02-06:
Thanks for the effort, today I had some time to try it. It works!! Has the CLI output to the HDMI, ethernet also works, it's got an IP. 
Now HA is prepping, so i think it'll be fine.

citruz, 2023-02-06:
That’s great, glad to hear!
My next goal is to build specific images for all the different boards of the Rock Pi 4 Family so that the model shows up correctly and the cpu frequencies are correct. Would be nice if you could test it on your 4B then.

closed 2023-02-06 by citruz

mthqwork, 2023-02-06:
Sure! Why not.

citruz, 2023-02-12:
I just published a beta release which includes an upgrade to a much more recent kernel version and specific images for the 4B and other boards.
Please give this release a try and let me know if it works correctly. It should now report the model correctly and fix the CPU frequency (1.8 Ghz).
https://github.com/citruz/haos-rockpi/releases/tag/9.5%2B20230212beta