citruz/haos-rockpi #2: Rock Pi 4C+

MBJ-M2BI, 2023-01-21:
### Describe the issue you are experiencing

can you create to rock pi 4C+ ?

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

can you create to rock pi 4C+ ?

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

can you create to rock pi 4C+ ?

### Anything in the Supervisor logs that might be useful for us?

```txt
can you create to rock pi 4C+ ?
```

### Anything in the Host logs that might be useful for us?

```txt
can you create to rock pi 4C+ ?
```

### System information

can you create to rock pi 4C+ ?

### Additional information

can you create to rock pi 4C+ ?

citruz, 2023-01-29:
Sorry for the delayed response, for some reason I did not get a notification.

Did you check if this build already works for the 4C+? In the HA forum, someone reported that he used this build successfully on an OKdo 4C+ (which seems to be the same as the RockPi 4C+).

https://community.home-assistant.io/t/home-assistant-on-rock-pi-4/477595/13

Please try to flash the existing image and let me know if it works.

MBJ-M2BI, 2023-01-29:
I tried.

And you are totally right. It just worked. Thank you for the system !
Do you know if there is any chance to change name on the OS ?

Like instead of Home Assistant, then Industry Assistant?

Venlig Hilsen/ Best Regards
      ***@***.***

Mike Boysen Jurock
CEO | M2BI
CVR DK-43754122
Mobile +45 26 88 33 60
Email ***@***.******@***.***>
Søtoften 15, 8740 Brædstrup, Denmark

https://community.home-assistant.io/t/home-assistant-on-rock-pi-4/477595/13

citruz, 2023-01-29:
That's great! Glad to hear that it works.

> Do you know if there is any chance to change name on the OS ?

In the web UI you mean? No official way that I am aware of. You would probably need to modify the core homeassistant package for that.

closed 2023-01-29 by citruz

MBJ-M2BI, 2023-01-29:
Yes totally nice work you made there.
I used hours to fin a solution for installing it, but then your software was there.

  *   PLING, it just worked.

So if I need to change the CORE will it then still update with new versions of HASSIO ?
Just so I not kill the access. My idea, is to setup up devices pre installed for other, and industry to help them collect data.

I have tried to contact Nabu Casa to ask them.
But as I see there is not any rights on the system, so I could just help other, in this case it could be nice if for industries, it was Industry Assistant, instead of Home Assistant.

Venlig Hilsen/ Best Regards
      ***@***.***

Mike Boysen Jurock
CEO | M2BI
CVR DK-43754122
Mobile +45 26 88 33 60
Email ***@***.******@***.***>
Søtoften 15, 8740 Brædstrup, Denmark

citruz, 2023-01-29:
Home Assistant is designed in a way so that every component is separated into its own Docker image/container. What you flash on your SD card is only the base system and the actual HA application will be downloaded as a container once the system has booted.
So to modify HA Core you would probably need to build a custom docker image and also provide a repo so that customers can update the image.
But I am definitely not an expert on HA customization and there might be a much better way to do this. You can try asking on the forum or discord.

MBJ-M2BI, 2023-01-29:
I will try to see if I can find a solution.
thank you very much ! 😊

Have a nice weekend.

Venlig Hilsen/ Best Regards
      ***@***.***

Mike Boysen Jurock
CEO | M2BI
CVR DK-43754122
Mobile +45 26 88 33 60
Email ***@***.******@***.***>
Søtoften 15, 8740 Brædstrup, Denmark

citruz, 2023-02-12:
I just published a beta release which includes an upgrade to a much more recent kernel version and specific images for the 4C+ and other boards.
Please give this release a try and let me know if it works. It should also improve hardware support on the 4C+ (other users had reported non-working USB ports).
https://github.com/citruz/haos-rockpi/releases/tag/9.5%2B20230212beta

MBJ-M2BI, 2023-02-14:
Just tried.

And works perfectly !
Thank you again.

Venlig Hilsen/ Best Regards
      ***@***.***

Mike Boysen Jurock
CEO | M2BI
CVR DK-43754122
Mobile +45 26 88 33 60
Email ***@***.******@***.***>
Søtoften 15, 8740 Brædstrup, Denmark

https://github.com/citruz/haos-rockpi/releases/tag/9.5%2B20230212beta