citruz/haos-rockpi #4: Rock Pi 4SE does not boot from emmc with haos. It does boot when haos on sd. Also boots from emmc with ubuntu.

KrickGregory, 2023-02-09:
### Describe the issue you are experiencing

Hey,

Getting crazy here. First of al many thanks for the build of haos for rockpi. I installed it on sd card first and everything went smooth and good. Today I bought an emmc card for performance BUT : 
rock 4se won't boot haos from emmc module. 
I tried : SPI bootstrap code can be disabled by connecting GPIO pins 23 with 25.
Same result. No boot.
If I flash ubuntu to emmc card everything works but then I have to install ha as docker and I don't like that.
So at the moment back to sd card but I have a heavy ha installation and think it won't take long to destroy the sd card.

Any idea of what could be the reason?

Kind regards,
Gregory

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

Home Assistant 2023.2.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Flash haos_rockpi-4b-plus-9.5.dev20230204 to emmc module with balena etcher and emmc to sd card module
2. Insert emmc module in rock 4se
3. Power on
4. No boot
...

### Anything in the Supervisor logs that might be useful for us?

```txt
no log because I can't ssh nor ping the device.
```

### Anything in the Host logs that might be useful for us?

```txt
no log because I can't ssh nor ping the device.
```

citruz, 2023-02-09:
This one will be tricky to debug without serial logs.
I don’t have a board with integrated SPI flash, so I can’t say for sure, but disabling the SPI by shorting the two pins should be correct. Whatever version/config of bootloader is on there might be incompatible with the HAOS image. But without the SPI flash, it _should_ work.
You could try flashing Ubuntu or armbian to the SD card, boot it and then flash the eMMC directly from the device. That is what I do since the 4B+ does not have removable eMMC.
Please also post the output of `lsblk` when you booted Ubuntu/armbian.

KrickGregory, 2023-02-09:
I'll give that a try tomorrow and let you know result. No HDMI indeed. Thanks

KrickGregory, 2023-02-10:
rock@rockpi-4b:~$ lsblk
NAME         MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
mmcblk0      179:0    0 119.1G  0 disk 
├─mmcblk0p1  179:1    0   3.9M  0 part 
├─mmcblk0p2  179:2    0     4M  0 part 
├─mmcblk0p3  179:3    0     4M  0 part 
├─mmcblk0p4  179:4    0   512M  0 part 
└─mmcblk0p5  179:5    0 118.6G  0 part /
mmcblk1      179:32   0  57.6G  0 disk 
├─mmcblk1p1  179:33   0    24M  0 part 
├─mmcblk1p2  179:34   0    24M  0 part 
├─mmcblk1p3  179:35   0   256M  0 part 
├─mmcblk1p4  179:36   0    24M  0 part 
├─mmcblk1p5  179:37   0   256M  0 part 
├─mmcblk1p6  179:38   0     8M  0 part 
├─mmcblk1p7  179:39   0    96M  0 part 
└─mmcblk1p8  179:40   0   1.3G  0 part 
mmcblk1boot0 179:64   0     4M  1 disk 
mmcblk1boot1 179:96   0     4M  1 disk 
mmcblk1rpmb  179:128  0     4M  0 disk 

rock@rockpi-4b:~$ df
Filesystem     1K-blocks    Used Available Use% Mounted on
udev             1940208       8   1940200   1% /dev
tmpfs             390620     528    390092   1% /run
/dev/mmcblk0p5 124267840 3118280 116147276   3% /
tmpfs            1953088       0   1953088   0% /dev/shm
tmpfs               5120       4      5116   1% /run/lock
tmpfs            1953088       0   1953088   0% /sys/fs/cgroup
tmpfs             390616       0    390616   0% /run/user/0
tmpfs             390616       0    390616   0% /run/user/1000

After flash from sd to emmc same result. No boot, no hdmi, no ... 

I'll buy a usb to ttl cable next week and will post log then.

citruz, 2023-02-10:
That output looks as expected, all the partitions on `mmcblk1` are there.
The last idea I have is that you could try overwriting mmcblk1boot0 and mmcblk1boot1 with zeros to make sure they contain no leftover bootloader which is used by the board.

citruz, 2023-02-12:
I just published a beta release which includes an upgrade to a much more recent kernel version and specific images for the 4SE and other boards.
Please give this release a try and let me know if it fixes the boot issue.
If the image for the 4SE does not boot from emmc either, please try the image for the 4C+. That board uses the same CPU variant as the 4SE and if the 4C+ image works, I might have an idea what the issue could be.

https://github.com/citruz/haos-rockpi/releases/tag/9.5%2B20230212beta

KrickGregory, 2023-02-13:
Hey,

First of all thanks for all the efforts. But although I wished something else I had no luck with booting from emmc with the new beta.
1. haos_rock-4se-9.5.dev20230212.img.xz on emmc with spi 23, 25 closed. No boot, no hdmi.
2. haos_rock-4se-9.5.dev20230212.img.xz on emmc with spi pins open. No boot, no hdmi. But I saw blue led winking twice.
3. haos_rockpi-4b-plus-9.5.dev20230204.img.xz on emmc with spi closed. No boot, no hdmi.
4. haos_rockpi-4b-plus-9.5.dev20230204.img.xz on emmc wit spi open. No boot, no hdmi.
None of the above made it possible to ping the board too.
5. haos_rock-4se-9.5.dev20230212.img.xz on sd card with spi pins open. Booting and ha is working. Restoring my backup at the moment.
I also did a mail to Radxa to ask what the problem could be. And they sent me a very clear answer : HAOS currently does not support ROCK 4.
:-)

Kind regards.

citruz, 2023-02-13:
Haha well, that is not a helpful answer considering that it works off the SD card.
Please try the 4C+ image as a final attempt. Those two boards share the same CPU variant.

KrickGregory, 2023-02-13:
Tried with 4C+ and seems to do something more. 
WIth spi pins open and closed I have same result. It seems to do something more because the blue heartbeat led is winking. But no hdmi or ping possible.
I have connected with a ttl cable following this guide https://wiki.radxa.com/Rockpi4/dev/serial-console but the only thing I get to see is a lot of questionmarks and all of a sudden it stops with questionmarks.

��������������������������������������������������������������������������������
��������������������������

Don't know if I can do something else. But keep in mind that I'm an electrician and not an IT guy. :-)

citruz, 2023-02-13:
The baudrate needs to be set to 115200 for these images. You should get some output then.

KrickGregory, 2023-02-13:
well, well, well, one step closer. With the 4plus image on emmc I don't have hdmi nor ping BUT in console with ttl I can log in
[  OK  ] Started Home Assistant CLI.                                            
[  OK  ] Started Serial Getty on ttyS2.                                         
[  OK  ] Reached target Login Prompts.                                          
[  OK  ] Reached target Multi-User System.                                      
                                                                                
Welcome to Home Assistant                                                       
homeassistant login: root                                                       
Last login: Wed Jul 13 11:34:31 on ttyS2                                        
Welcome to Home Assistant OS.                                                   
                                                                                
Use `ha` to access the Home Assistant CLI.                                      
# ha os info                                                                    
board: rock-4c-plus                                                             
boot: A                                                                         
data_disk: /dev/mmcblk1p8                                                       
update_available: false                                                         
version: 9.5.dev20230212                                                        
version_latest: null                                                            
#

citruz, 2023-02-13:
Great! You should be able to get your IP using `ip a` or `ha network info` and connect to it.

KrickGregory, 2023-02-13:
Ok, I'll give it a try. But first reflash to the plus on my emmc because to give you some more information I flashed the beta 4se. It seems that it keeps booting all the time. Just for your information this is what the log gave me. 

I'll be back :-)

MMC read: dev # 0, block # 1214464, count 64 ... 64 blocks read: OK
loading environment from haos-config.txt
65 bytes read in 7 ms (8.8 KiB/s)
59 bytes read in 8 ms (6.8 KiB/s)
Loading standard device tree rk3399-rock-4se.dtb
60434 bytes read in 19 ms (3 MiB/s)
Trying to boot slot A, 0 attempts remaining. Loading kernel ...
39735808 bytes read in 3108 ms (12.2 MiB/s)
storing env...

MMC write: dev # 0, block # 1214464, count 64 ... 64 blocks written: OK
Starting kernel with args zram.enabled=1 zram.num_devices=3 apparmor=1 security=apparmor systemd.machine_id=08fe15ef04de40a687ea3578675a169d fs
ck.repair=yes root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd rootfstype=squashfs ro rootwait rauc.slot=A earlycon=uart8250,mmio32,0xff1a000
0 console=ttyS2,115200n8
Moving Image from 0x2080000 to 0x2200000, end=4890000
## Flattened Device Tree blob at 01f00000
   Booting using the fdt blob at 0x1f00000
   Loading Device Tree to 00000000f1ea3000, end 00000000f1f1afff ... OK

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.1.11 (builder@dfa07d100059) (aarch64-buildroot-linux-gnu-gcc.br_real (Buildroot -g14dcc6f7) 9.4.0, GNU ld (GNU B
inutils) 2.36.1) #1 SMP PREEMPT Sun Feb 12 16:23:01 UTC 2023
[    0.000000] Machine model: Radxa ROCK 4SE
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000ff1a0000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] efi: UEFI not found.
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] NUMA: Faking a node at [mem 0x0000000000200000-0x00000000f7ffffff]
[    0.000000] NUMA: NODE_DATA [mem 0xf77e0900-0xf77e2fff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x00000000f7ffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x00000000f7ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x00000000f7ffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] cma: Reserved 32 MiB at 0x00000000efe00000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 21 pages/cpu s45480 r8192 d32344 u86016
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Fallback order for Node 0: 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 999432
[    0.000000] Policy zone: DMA
[    0.000000] Kernel command line: zram.enabled=1 zram.num_devices=3 apparmor=1 security=apparmor systemd.machine_id=08fe15ef04de40a687ea35786
75a169d fsck.repair=yes root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd rootfstype=squashfs ro rootwait rauc.slot=A earlycon=uart8250,mmio32
,0xff1a0000 console=ttyS2,115200n8
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 3910416K/4061184K available (17792K kernel code, 3796K rwdata, 9344K rodata, 7744K init, 637K bss, 118000K reserved, 327
68K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=6, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=6.
[... 484 lines cut ...]
256B stride
lpddr4_set_rate: change freq to 800MHz 1, 0
Trying to boot from BOOTROM
Returning to boot ROM...

U-Boot SPL 2022.10 (Feb 12 2023 - 16:17:15 +0000)
Trying to boot from MMC2
NOTICE:  BL31: v2.8(release):0f2d0e5
NOTICE:  BL31: Built : 16:16:52, Feb 12 2023

U-Boot 2022.10 (Feb 12 2023 - 16:17:15 +0000)

Model: Radxa ROCK Pi 4B
DRAM:  3.9 GiB
PMIC:  RK808 
Core:  282 devices, 29 uclasses, devicetree: separate
MMC:   mmc@fe310000: 2, mmc@fe320000: 1, mmc@fe330000: 0
Loading Environment from MMC... *** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Model: Radxa ROCK Pi 4B
Net:   eth0: ethernet@fe300000
Hit any key to stop autoboot:  2

KrickGregory, 2023-02-13:
seems to be something with  network or so. Log with the 4plus image 

 
Use `ha` to access the Home Assistant CLI.
# ha network info
Error: System is not ready with state: setup
# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: docker0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue 
    link/ether 02:42:be:64:fa:f6 brd ff:ff:ff:ff:ff:ff
    inet 172.30.232.1/23 brd 172.30.233.255 scope global docker0
       valid_lft forever preferred_lft forever
    inet6 fe80::42:beff:fe64:faf6/64 scope link 
       valid_lft forever preferred_lft forever
4: vethef01077@if3: <BROADCAST,MULTICAST,UP,LOWER_UP,M-DOWN> mtu 1500 qdisc noqueue master docker0 
    link/ether e2:2c:d6:db:e6:0f brd ff:ff:ff:ff:ff:ff
    inet6 fe80::e02c:d6ff:fedb:e60f/64 scope link 
       valid_lft forever preferred_lft forever
5: hassio: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue 
    link/ether 02:42:db:4d:95:53 brd ff:ff:ff:ff:ff:ff
    inet 172.30.32.1/23 brd 172.30.33.255 scope global hassio
       valid_lft forever preferred_lft forever
    inet6 fe80::42:dbff:fe4d:9553/64 scope link 
       valid_lft forever preferred_lft forever
7: vethaac6a6b@if6: <BROADCAST,MULTICAST,UP,LOWER_UP,M-DOWN> mtu 1500 qdisc noqueue master hassio 
    link/ether 32:0a:c7:e1:6e:51 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::300a:c7ff:fee1:6e51/64 scope link 
       valid_lft forever preferred_lft forever
9: veth19532b0@if8: <BROADCAST,MULTICAST,UP,LOWER_UP,M-DOWN> mtu 1500 qdisc noqueue master hassio 
    link/ether 32:45:71:25:58:3e brd ff:ff:ff:ff:ff:ff
    inet6 fe80::3045:71ff:fe25:583e/64 scope link 
       valid_lft forever preferred_lft forever
11: veth8ccb5ee@if10: <BROADCAST,MULTICAST,UP,LOWER_UP,M-DOWN> mtu 1500 qdisc noqueue master hassio 
    link/ether e6:1d:23:b0:84:3b brd ff:ff:ff:ff:ff:ff
    inet6 fe80::e41d:23ff:feb0:843b/64 scope link 
       valid_lft forever preferred_lft forever
13: veth8d32184@if12: <BROADCAST,MULTICAST,UP,LOWER_UP,M-DOWN> mtu 1500 qdisc noqueue master hassio 
    link/ether 72:90:af:a3:d3:d5 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::7090:afff:fea3:d3d5/64 scope link 
       valid_lft forever preferred_lft forever
15: veth570450f@if14: <BROADCAST,MULTICAST,UP,LOWER_UP,M-DOWN> mtu 1500 qdisc noqueue master hassio 
    link/ether 32:d8:9c:cd:fe:08 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::30d8:9cff:fecd:fe08/64 scope link 
       valid_lft forever preferred_lft forever

citruz, 2023-02-13:
Hm `eth0` is missing completely. Can you check `dmesg` if there are any errors related to network (or other errors)?

Just to make sure I understand it correctly: When you boot from SD card, network is working but if you boot the same image from eMMC, it does not work?

KrickGregory, 2023-02-13:
Yes, you are right. On sd card everything goes perfect without doing anythin. But on emmc not. I know strange.

here is the ouput of dmes but it's a lot sorry 

# dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.1.11 (builder@8dc595c5f976) (aarch64-buildroot-linux-gnu-gcc.br_real (Buildroot -g14dcc6f7) 9.4.0, GNU ld (GNU Binuti
ls) 2.36.1) #1 SMP PREEMPT Sun Feb 12 19:47:22 UTC 2023
[    0.000000] Machine model: Radxa ROCK 4C+
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000ff1a0000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] efi: UEFI not found.
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] NUMA: Faking a node at [mem 0x0000000000200000-0x00000000f7ffffff]
[    0.000000] NUMA: NODE_DATA [mem 0xf77e3900-0xf77e5fff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x00000000f7ffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x00000000f7ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x00000000f7ffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] cma: Reserved 32 MiB at 0x00000000efe00000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 21 pages/cpu s45480 r8192 d32344 u86016
[    0.000000] pcpu-alloc: s45480 r8192 d32344 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 [0] 4 [0] 5 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Fallback order for Node 0: 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 999432
[    0.000000] Policy zone: DMA
[    0.000000] Kernel command line: zram.enabled=1 zram.num_devices=3 apparmor=1 security=apparmor systemd.machine_id=e25a88e7b0bf44b8bbaeef2008d470
98 fsck.repair=yes root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd rootfstype=squashfs ro rootwait rauc.slot=A earlycon=uart8250,mmio32,0xff1a000
0 console=ttyS2,115200n8
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 3910428K/4061184K available (17792K kernel code, 3796K rwdata, 9344K rodata, 7744K init, 637K bss, 117988K reserved, 32768K c
ma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=6, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=6.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=6
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 256 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: GICv3 features: 16 PPIs
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000fef00000
[    0.000000] ITS [mem 0xfee20000-0xfee3ffff]
[    0.000000] ITS@0x00000000fee20000: allocated 65536 Devices @480000 (flat, esz 8, psz 64K, shr 0)
[    0.000000] ITS: using cache flushing for cmd queue
[    0.000000] GICv3: using LPI property table @0x0000000000440000
[    0.000000] GIC: using cache flushing for LPI property table
[    0.000000] GICv3: CPU0: using allocated LPI pending table @0x0000000000450000
[    0.000000] GICv3: GIC: PPI partition interrupt-partition-0[0] { /cpus/cpu@0[0] /cpus/cpu@1[1] /cpus/cpu@2[2] /cpus/cpu@3[3] }
[... 630 lines cut ...]
[  151.599811] audit: type=1334 audit(1657711994.348:230): prog-id=70 op=LOAD
[  151.600775] audit: type=1334 audit(1657711994.352:231): prog-id=71 op=LOAD
[  151.601089] audit: type=1334 audit(1657711994.352:232): prog-id=72 op=LOAD
[  151.659958] audit: type=1334 audit(1657711994.408:233): prog-id=73 op=LOAD
[  151.660100] audit: type=1300 audit(1657711994.408:233): arch=c00000b7 syscall=280 success=yes exit=15 a0=5 a1=40001c3940 a2=78 a3=0 items=0 ppid=
3206 pid=3216 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="runc" exe="/usr/bin/runc" subj
=unconfined key=(null)
[  151.660188] audit: type=1327 audit(1657711994.408:233): proctitle=72756E63002D2D726F6F74002F7661722F72756E2F646F636B65722F72756E74696D652D72756E6
32F6D6F6279002D2D6C6F67002F7661722F72756E2F646F636B65722F636F6E7461696E6572642F6461656D6F6E2F696F2E636F6E7461696E6572642E72756E74696D652E76322E74617
36B2F6D6F62792F35393065303935
[... cut ...]
[  181.729082] kauditd_printk_skb: 17 callbacks suppressed
[  181.729096] audit: type=1334 audit(1657712024.480:245): prog-id=26 op=UNLOAD
[  181.729120] audit: type=1334 audit(1657712024.480:246): prog-id=25 op=UNLOAD
[  181.729138] audit: type=1334 audit(1657712024.480:247): prog-id=24 op=UNLOAD
[  182.353345] audit: type=1334 audit(1657712025.104:248): prog-id=72 op=UNLOAD
[  182.353378] audit: type=1334 audit(1657712025.104:249): prog-id=71 op=UNLOAD
[  182.353396] audit: type=1334 audit(1657712025.104:250): prog-id=70 op=UNLOAD

citruz, 2023-02-13:
> here is the ouput of dmes but it's a lot sorry

You can wrap it in triple backticks to make it a bit more readable.

```
like this
```

Interesting: 

> [ 134.355310] eth0: renamed from veth4feb443

Looks like it is there... can you try bringing it down and up again:

```
ip a sh dev eth0
ip link set eth0 down
ip link set eth0 up
ip a sh dev eth0
```

citruz, 2023-02-13:
These log lines seem to be the problem. It means that the device driver could not find or initialize the hardware.
```
[ 17.316552] platform fe300000.ethernet: deferred probe pending
[ 17.316575] platform ff770000.syscon:usb2phy@e460: deferred probe pending
[ 17.316585] platform fe320000.mmc: deferred probe pending
[ 17.316594] platform ff320000.syscon:io-domains: deferred probe pending
[ 17.316604] platform ff940000.hdmi: deferred probe pending
[ 17.316612] platform ff770000.syscon:io-domains: deferred probe pending
[ 17.316621] platform usb@fe800000: deferred probe pending
[ 17.316674] platform fe900000.usb: deferred probe pending
[ 17.316684] platform cpufreq-dt: deferred probe pending
```

I don't really understand why the behavior is different between SD and eMMC boot though. Maybe it has something to do with the low level initialization in the boot loader.
Can you please post the U-Boot logs (until `[ 0.000000] Booting Linux on physical CPU`) for the eMMC and SD card boot,  with the GPIO pins shorted and not shorted?
Please use "````" for the logs and indicate for all 4 logs whether the ethernet device was detected in that boot or not.

KrickGregory, 2023-02-14:
Goodmorning,

Did what you asked but didn't understand "Please use "````" for the logs" so I attach 4 complete logs in here. Hope it helps you.
SD with spi open -> ping to rock ok
SD with spi closed -> pint to rock ok
EMMC with spi open and closed -> no ping to the card possible.
Little remarque. I left the emmc on the card when logging with ssd because it just boots ok.
[LogEMMCspiClosed.txt](https://github.com/citruz/haos-rockpi/files/10729703/LogEMMCspiClosed.txt)
[LogEMMCspiOpen.txt](https://github.com/citruz/haos-rockpi/files/10729704/LogEMMCspiOpen.txt)
[LogSDspiClosed.txt](https://github.com/citruz/haos-rockpi/files/10729705/LogSDspiClosed.txt)
[LogSDspiOpen.txt](https://github.com/citruz/haos-rockpi/files/10729706/LogSDspiOpen.txt)

citruz, 2023-02-14:
Thanks for uploading the logs. I will have a look later today.

citruz, 2023-02-14:
Very interesting. First of all, the pins open vs closed does not seem to make a difference as to what bootloader is used so we can ignore that.

What is really throwing me off is that even in the logs where it finally came up, there are a lot of kernel errors and even panics. In [LogSDspiOpen.txt](https://github.com/citruz/haos-rockpi/files/10729706/LogSDspiOpen.txt), it reboots three times before you get to the login prompt.

The panics look like this:
```
[    3.006263] mmc1: running CQE recovery                                       
[    3.010927] ------------[ cut here ]------------                             
[    3.016095] mmc1: cqhci: spurious TCN for tag 4                              
[    3.021243] WARNING: CPU: 0 PID: 95 at drivers/mmc/host/cqhci-core.c:786 cqhc
i_irq+0x38c/0x590 
```
There are also other IO errors:
```
[    4.817637] I/O error, dev mmcblk1, sector 250154 op 0x0:(READ) flags 0x800 p
hys_seg 22 prio class 2                                                         
[    4.828041] SQUASHFS error: Failed to read block 0x4125614: -5  
```

`mmcblk1` is referring to the eMMC. Did you leave the eMMC attached while booting from SD card? That could cause problems and it may even have loaded the root filesystem from there, depending on which device was initialized first. Please always have only either the SD or eMMC connected to make sure there is no interference.

I found a post on the Radxa forum where someone is experiencing the same kernel panic, also on a 4SE. Maybe it is common problem with eMMC on the 4SE? What power supply are you using?
https://forum.radxa.com/t/rock4-se-emmc-input-output-errors-read-only-fs/14406

The eMMC logs don't have any errors, but there the hardware is not detected.

What would be interesting would be a log of the 4SE image booted from eMMC without the SD card inserted. 

Unfortunately this is very hard to debug for me without access to the hardware. The 4SE is also the only board where Radxa does not provide a dedicated device tree, so people have to use the images meant for other boards such as the 4B or 4C+. But of course these can have slight incompatibilities.

citruz, 2023-02-14:
To check whether these kernel errors also occur with distros other than the Radxa Ubuntu, you could try armbian: https://www.armbian.com/rockpi4/
They recommend to use the 4B image for the 4SE. Check whether it boots from eMMC and if there are any errors.

citruz, 2023-02-14:
This looks also very similar: https://forum.armbian.com/topic/20002-nanopc-t4-new-kernel-2202-generates-issues-on-mmc2-and-makes-system-not-properly-working/
No network and same error messages. Sounds like they have tracked it down to some kernel changes in 5.x. Maybe the Radxa Ubuntu works because it uses an old 4.4 kernel (at least I think it does)?

KrickGregory, 2023-02-15:
With ubuntu 4.4 kernel it boots from emmc and I can ssh into it. Couldn't take log because all strange characters even if I set baudrate to 1500000.
With 4se image it boots and ha starts but with a lot of errors. See logfile.

To me it seems a problem with the kernel above 4.4 for some strange reason.
I will by this ssd 
https://www.coolblue.be/nl/product/915798/kingston-nv2-pcie-4-0-nvme-ssd-500gb.html?cmt=c_a%2Ccid_18623141970%2Caid_142142284003%2Ctid_aud-918309158136%3Apla-327133878214%2Cgn_g%2Cd_c&gclid=CjwKCAiA_6yfBhBNEiwAkmXy55RhJkd2Y6vKUOYOyIJt224mr8LvJZq0eS8qN56U3T3ahSMz0sDqoBoCEM0QAvD_BwE
and boot from sd card and then move ha to data ssd. Hope that will work. What do you think?

I realy want to thank you for looking into this problem.

[LogEMMC4se.txt](https://github.com/citruz/haos-rockpi/files/10740149/LogEMMC4se.txt)

Elarion245, 2023-05-22:
I am also thinking about migrating from an sd card to emmc. Sadly the rock 4c+ has no m2 connector, so ssd is not possible to only emmc.

Based on the above: would this be expected to work? Can I go ahead and buy an emmc or not yet until a new kernel is released?

citruz, 2023-05-24:
The problem above feels like it is a hardware issue of that specific board or that specific eMMC module, but that is only a guess from my side. It _should_ work 🤷‍♂️

Elarion245, 2023-05-24:
Very good. I'll go ahead and buy one then once I find a good offer. Will report back then on this in a few weeks.

citruz, 2023-05-25:
Just wondering, since it came up in another issue, doesn't the 4C+ have an M.2 connector on the top? For a half-sized ssd
https://jamesachambers.com/rock-pi-4c-plus-w-nvme-sbc-review/

Elarion245, 2023-05-25:
I just found that it is not about the Pi in the name. It is about the revision of the board. The M2 was removed when upgrading the design from v1.2 to v1.4: https://wiki.radxa.com/Rockpi4/hardware/revision

Elarion245, 2023-06-08:
I can confirm that the EMMC is working fine on the ROCK 4C+ (new version 1.4 without M.2) slot. 
I had to flash it three times though, as the first time the EMMC wasn't properly connected to the microSD adapter. Second attemt was fine on flashing, but I did a hard power down and afterwards it also didn't come up again. Third time I flashed, changed the haos-config, then reassembled everything and finally did the power on at its proper place in my apartment.

The overlay which disables the LED is also helpful here, as it indicates any bootup problems (e.g. if the green light remains on despite activating the overlay to disable it).

My next goal is now to flash Debian on the microSD. This way I should then be able to remotely connect from my laptop for any future EMMC updates without having to disassemble everything...

Elarion245, 2023-06-12:
A short summary on my Dual-Boot attempts:
With HOAS on EMMC, I've tried the following:

- Debian on SD -> does not boot. Black screen after uboot
- Ubunu on SD -> same, black after uboot
- Manjaro cannot be downloaded at the moment (404)
- DiePi on SD -> boots! However due to kernel 5.15 WIFI does not work (OOB interrupts not disabled). Also only one USB-Port worked, so I had to use a small HUB on the one port. 

Apart from that I could successfully add MATE and xRDP on top. This way I can now simply insert the microSD for DietPi to boot.
Whenever the next HAOS release will be available, I'll try updating the EMMC via DietPi then instead of manually unplugging the EMMC and putting it into the microSD adapter. Hopefully this WIFI and USB but will be fixed in the future on DietPi. For the time being however I am happy that the DualBoot worked with at least one system. I have not tried others afterwards then.

citruz, 2024-05-02:
Closing due to inactivity

closed 2024-05-02 by citruz