citruz/haos-rockpi #5: Only one USB port working

magjo, 2023-02-10:
### Describe the issue you are experiencing

When using haos_rockpi-4b-plus-9.5.dev20230129.img only one out of four USB ports on my OKdo Rock 4C+ is working (the upper USB 3 (blue) port). I discovered the problem when trying to connect a RFXtrx tranceiver, but it's the same with an ordinary USB keyboard. If I switch to another SD card with Ubuntu installed, all four USB ports are working when trying with a keyboard.

Is there anything I could have done wrong?

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

9.5.dev20230129

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Connect USB keyboard to an USB port
2. Try to use the keybord - nothing will happen for three out of for USB ports (nothing happens when you press keys, num lock indicator does not light up)

### Anything in the Supervisor logs that might be useful for us?

```txt
I cannot find a way to select any log provider on the linked page.
```

### Anything in the Host logs that might be useful for us?

```txt
I cannot find a way to select any log provider on the linked page.
```

MrDadoo, 2023-02-10:
Rock Pi images often comes with the upper blue USB port set to otg mode.

A change in /boot/hw_intfc.conf controls this.

See this doc
 https://wiki.radxa.com/Rockpi4/dev/usb-host-and-otg-switch-on-rock-pi-4

(Though, I didn't notice this on this HAOS image.)

magjo, 2023-02-10:
Maybe I did not understand correctly, but the upper blue USB port is the only one working. The lower blue USB port and both black USB ports are not working. Which ports are affected by this setting?

MrDadoo, 2023-02-10:
> Maybe I did not understand correctly, but the upper blue USB port is the only one working. The lower blue USB port and both black USB ports are not working. Which ports are affected by this setting?

I’m sorry. I missunderstood.

magjo, 2023-02-12:
> This may be caused by the fact that the image uses the 4B+ device tree which could cause such hardware issues. My next goal is to produce specific images for the different rock pi boards which use the correct hardware definitions.

Sounds good. At the moment I only have one USB device to connect, so it works ok right now, but it would be nice to be able to use all USB ports in the future if I would like to add another device (like a Z-wave stick or similar).

I would also like to give you my warmest thanks for carrying on with this projekt. It is highly appreciated!

citruz, 2023-02-12:
I just published a beta release which includes an upgrade to a much more recent kernel version and specific images for the 4C+ and other boards.
Please give this release a try and let me know if it fixes the hardware issues.
https://github.com/citruz/haos-rockpi/releases/tag/9.5%2B20230212beta

magjo, 2023-02-13:
That's quick! Big thanks!

I have quickly tested the image haos_rock-4c-plus-9.5.dev20230212 and all USB ports seems to be working, at least with a keyboard. I have just booted the image with a screen and a keyboard connected and verified that I can type commands at the prompt, for all four USB ports. I have not testad that Home Assistant is working.

Is there something you would like me to test to help you verifying that the image is working as intendet?

citruz, 2023-02-13:
Great, glad that it works now! Please also check if `lscpu` now reports the correct CPU frequency of 1.5 GHz.

magjo, 2023-02-13:
Where do I enter that command? At the `ha >` prompt?

citruz, 2023-02-13:
Enter “login” at the ha prompt to get into the Linux shell and then “lscpu”

MrDadoo, 2023-02-13:
> Enter “login” at the ha prompt to get into the Linux shell and then “lscpu”

What credentials should be used?

magjo, 2023-02-13:
It says 1512 MHz

[image]

citruz, 2023-02-13:
> > Enter “login” at the ha prompt to get into the Linux shell and then “lscpu”
> 
> What credentials should be used?

None are required if you go through `ha`. Otherwise the username is root without any password.

closed 2023-02-13 by citruz

MrDadoo, 2023-02-13:
> None are required if you go through `ha`. Otherwise the username is root without any password.

I tried through the ssh add-on. There I get askes for credentials but neither root nor my username works.

citruz, 2023-02-13:
The SSH add-on does not give you "real" access to the host system, you will be dropped in an isolated docker container. If you want real ssh access follow this guide: https://developers.home-assistant.io/docs/operating-system/debugging/