citruz/haos-rockpi #7: RockPi 4B+ does not boot from sd-card automatically

christschnman, 2023-02-26:
### Describe the issue you are experiencing

Hi, I don't think that is a bug, but maybe you can help me with your RockPi-skills for this problem:

I use a RockPi 4B+ from an preconfigured device (Nebra Helium Miner) with soldered eMMC Storage. I want to keep this configuration for all cases, so I installed your Image only on the SD-Card.  The Problem is that it doesn't boot from the sd card by itself. When I press the middle from the three onboard buttons ("Maskrom") during startup it works fine. 
Is there a way to boot from card permanently. Is there a Bios for this setting? I have found something about bridge the two gpio-pins 23 and 25, but I don't know enough about it to test it.

### What operating system image do you use?

rock-4c-plus

### What version of Home Assistant Operating System is installed?

6.6

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
?
```

### Anything in the Host logs that might be useful for us?

```txt
?
```

citruz, 2023-02-26:
I don’t think this is possible, unfortunately. The Rock Pi always tries to boot from eMMC first. Only if it doesn’t find a bootloader on eMMC, it will try to boot from SD.
Shorting GPIO pins 23 and 25 only applies for models that have a built in SPI flash and allows skipping boot from SPI. The 4B+ doesn’t have SPI flash and I am not aware that a similar way exists to skip eMMC.

christschnman, 2023-03-04:
Thank you for your advice,

Do you think it would be possible to backup the eMMC memory and then install the miniloader according to your documentation? As I said, I would like to keep the old configuration just in case.

Or would it be possible with simple means to edit the existing bootloader on the eMMC in order to boot from the SD.

citruz, 2023-03-04:
The miniloader is only 8MB so you could make a backup of the first 8MB of eMMC (or some more, just to be sure) and restore it later.
Editing the existing bootloader is not possible unfortunately.

closed 2023-04-23 by citruz