citruz/haos-rockpi #10: Home Assistant OS 10.0 20230423 release for rock-4-se is missing

JoeMoer222, 2023-05-03:
### Describe the issue you are experiencing

 release for rock-4-se is missing

### What operating system image do you use?

rock-4se

### What version of Home Assistant Operating System is installed?

na

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1.
2.
3.
...
NA

### Anything in the Supervisor logs that might be useful for us?

```txt
na
```

### Anything in the Host logs that might be useful for us?

```txt
na
```

### Additional information

If I use the image for rock-4-b on the rock-4-se there is no wifi in Home Assistant.

citruz, 2023-05-04:
Ooops sorry, I accidentally removed the Rock 4SE from the build. Will fix asap and push the missing image.

citruz, 2023-05-04:
https://github.com/citruz/haos-rockpi/releases/download/10.0%2B20230423/haos_rock-4se-10.0.dev20230504.img.xz

Let me know if this fixes the wifi issue.

closed 2023-05-04 by citruz

frecl, 2023-05-08:
Using this image om my 4SE I can at least list the wifi-networks but I am using a wired ethernet connection myself.