citruz/haos-rockpi #11: Rock Pi 4B WiFi not working

borconi, 2023-05-15:
### Describe the issue you are experiencing

Fresh install on a Pi4B, however WiFi module doesn't work. Can't provide the rest of the required info as there is no lan, so needed to configure it over wifi from cmd line first.

lsmod doesn't seem to show the ap6256 module loaded....

ha network info reports wifi as null so does nmcli.

### What operating system image do you use?

rockpi-4b

### What version of Home Assistant Operating System is installed?

10.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Flash new images to SD card.

### Anything in the Supervisor logs that might be useful for us?

```txt
#N/A
```

### Anything in the Host logs that might be useful for us?

```txt
#N/A
```

### System information

#N/A

### Additional information

#N/A

borconi, 2023-05-15:
Rolled back to 9.5 (20230204 version) and there wifi works as expected. Suspecting it's the new kernel which has broken the wifi driver.

citruz, 2023-05-19:
Sorry to hear that it broke. Your remark about the kernel upgrade sounds likely to be true. It's just weird that it works on my 4B+ but maybe that board has a different Wifi chipset. It's hard to find detailed specifications on the radxa site and I currently don't have access to mine.
I will look into it next week.

borconi, 2023-05-19:
Same WIFI chip, the AP6256 it's used on both, so that makes it even more interesting.... will give the B+ image a test just for curiosity.

citruz, 2023-05-21:
Hm so it definitely works on the 4B+:
```
# uname -a
Linux homeassistant 6.1.24 #1 SMP PREEMPT Mon Apr 24 00:25:13 UTC 2023 aarch64 HAOS
# readlink /sys/class/net/wlan0/device/driver/module
../../../../module/brcmfmac
# ha network scan wlan0
accesspoints:
- frequency: 5240
  mac: *****
  mode: infrastructure
  signal: 70
  ssid: *****
```

Can you please
1. post the output of `readlink /sys/class/net/wlan0/device/driver/module` on your working 5.x installation
2. check the boot log on 6.x for any errors related to brcmfmac or a related driver? (either looking at the serial output or with journalctl)

for comparison, this it what it looks like on my board:
```
May 21 17:51:47 homeassistant kernel: brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43456-sdio for chip BCM4345/9
...
May 21 17:51:47 homeassistant kernel: brcmfmac mmc2:0001:1: Direct firmware load for brcm/brcmfmac43456-sdio.radxa,rockpi4b-plus.bin failed with error -2
...
May 21 17:51:47 homeassistant kernel: brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/9 wl0: May 14 2020 17:26:08 version 7.84.17.1 (r871554) FWID 01-3d9e1d87
```
Looks like the firmware load error is expected or at least not critical.

borconi, 2023-05-21:
I'll try to do it tomorrow (if I don't forget, just to find )

Emil Borconi Szedressy
Sent from my mobile device, please excuse any typos

borconi, 2023-05-22:
On 10.x  there is no `/sys/class/net/device` probably because the driver failed to load (sorry not having any wired network in the office so posting pic instead of text as I can't ssh in and too lazy to copy. This is with both 4B and 4B+ images on 10.0 version.

Screenshot of error: https://photos.app.goo.gl/z2xeEmAfKRv9KgRt9

borconi, 2023-05-22:
OK a very interesting observation..... I left the system running for a longer time, without doing anything on it, busy with other stuff. And now there is a WLAN. NetworkManager found the WiFi aprox 2 minutes after system completed booting.... and scanning is working as well.  I guess issue can be closed... albeit not sure why the bring up of the wlan0 is so late, much later then system is up....

citruz, 2023-05-22:
Yes from the error messages it looked like a timeout when the kernel was trying to communicate with the chipset controller. Maybe some hardware problem which causes instability. It some point the kernel tries again and it works the second time.
Not great, but also not much I could do about that anyway.
Let me know if the wifi is stable once it has been set up.

closed 2023-05-22 by citruz

Elarion245, 2023-05-22:
I also have the issue of no wifi on the 4c+. During boot I get via dmesg:

[ 6.991541] brcmfmac mmc2:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.radxa,rock-4c-plus.bin failed with error -2
[ 9.867500] ieee80211 phy0: brcmf_bus_started: failed: -110
[ 9.873962] ieee80211 phy0: brcmf_attach: dongle is not responding: err=-110

I have yet to check if the device appears later after finishing the boot up. Also I have not checked yet the prior version. The 110 error sounds like some sort of power saving feature?

Please don't close this topic yet.

citruz, 2023-05-23:
There are multiple threads in the upstream repo on non-working wifi or bluetooth after the upgrade to HAOS 10. These also seem to experience some timeout issues when talking to the controller:
https://github.com/home-assistant/operating-system/issues/2485#issuecomment-1514708604
https://github.com/home-assistant/operating-system/issues/2535#issuecomment-1537530910

citruz, 2023-05-23:
This looks very relevant:
https://www.spinics.net/lists/stable/msg643721.html
It's the exact same 4 errors as in @borconi 's screenshot. I don't think the patch has landed in 6.1 yet.

Elarion245, 2023-05-23:
Good catch! This indeed sounds like the root cause - and is explaining the suspicion concerning power saving...
So hopefully this will be resolved with the next release as well.

citruz, 2023-05-24:
I just published a beta release which upgrades the kernel to 6.1.29.
While I am not super confident that this will fix the issue, you can still try regardless.
https://github.com/citruz/haos-rockpi/releases/tag/10.1%2B20230523beta

Elarion245, 2023-05-24:
Tests finished. Wifi still results in:
[    9.858932] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[    9.865524] ieee80211 phy0: brcmf_bus_started: failed: -110
[    9.871818] ieee80211 phy0: brcmf_attach: dongle is not responding: err=-110
[    9.895380] brcmfmac: brcmf_sdio_firmware_callback: brcmf_attach failed

But: I was trying to get my Xiaomi Bluetooth BLE Kettle integrated for some days and failed. With the new version it suddenly worked with the Passive BLE add-on. So apparently this was a kernel issue as well.

Do you have any idea when the WiFi fix will be integrated into the kernel?

Elarion245, 2023-05-24:
Just for your info, I attach here the complete dmesg of the 4C+ with the new beta:

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.1.29 (builder@86fa23a482bc) (aarch64-buildroot-linux-gnu-gcc.br_real (Buildroot -gc00352ed6) 11.3.0, GNU ld (GNU Binutils) 2.38) #1 SMP PREEMPT Tue May 23 22:47:11 UTC 2023
[    0.000000] Machine model: Radxa ROCK 4C+
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000ff1a0000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] efi: UEFI not found.
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] NUMA: Faking a node at [mem 0x0000000000200000-0x00000000f7ffffff]
[    0.000000] NUMA: NODE_DATA [mem 0xf77d8380-0xf77dafff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x00000000f7ffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x00000000f7ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x00000000f7ffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] cma: Reserved 32 MiB at 0x00000000efe00000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 21 pages/cpu s45480 r8192 d32344 u86016
[    0.000000] pcpu-alloc: s45480 r8192 d32344 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 [0] 4 [0] 5
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Fallback order for Node 0: 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 999432
[    0.000000] Policy zone: DMA
[    0.000000] Kernel command line: zram.enabled=1 zram.num_devices=3 net.naming-scheme=v250 systemd.machine_id=1d515f7c5ac24e0285afcf9bcc6333d3 fsck.repair=yes  root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd rootfstype=squashfs ro rootwait rauc.slot=A earlycon=uart8250,mmio32,0xff1a0000 console=ttyS2,115200n8
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 3910168K/4061184K available (17920K kernel code, 3804K rwdata, 9336K rodata, 7808K init, 638K bss, 118248K reserved, 32768K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=6, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=6.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=6
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 256 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: GICv3 features: 16 PPIs
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000fef00000
[    0.000000] ITS [mem 0xfee20000-0xfee3ffff]
[    0.000000] ITS@0x00000000fee20000: allocated 65536 Devices @480000 (flat, esz 8, psz 64K, shr 0)
[    0.000000] ITS: using cache flushing for cmd queue
[    0.000000] GICv3: using LPI property table @0x0000000000440000
[    0.000000] GIC: using cache flushing for LPI property table
[    0.000000] GICv3: CPU0: using allocated LPI pending table @0x0000000000450000
[    0.000000] GICv3: GIC: PPI partition interrupt-partition-0[0] { /cpus/cpu@0[0] /cpus/cpu@1[1] /cpus/cpu@2[2] /cpus/cpu@3[3] }
[    0.000000] GICv3: GIC: PPI partition interrupt-partition-1[1] { /cpus/cpu@100[4] /cpus/cpu@101[5] }
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000001] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.013839] Console: colour dummy device 80x25
[... 833 lines cut ...]
[  335.141543] audit: type=1327 audit(1684928837.954:387): proctitle=61707061726D6F725F706172736572002D2D72656D6F7665002D2D77726974652D6361636865002D2D63616368652D6C6F63002F6D6E742F646174612F73757065727669736F722F61707061726D6F722F6361636865002F6D6E742F646174612F73757065727669736F722F61707061726D6F722F32616434633733615F6861
[  469.206347] audit: type=1334 audit(1684928972.018:388): prog-id=124 op=LOAD
[  469.209596] audit: type=1334 audit(1684928972.022:389): prog-id=125 op=LOAD
[... cut ...]
[  511.755492] audit: type=1327 audit(1684929014.566:398): proctitle=72756E63002D2D726F6F74002F7661722F72756E2F646F636B65722F72756E74696D652D72756E632F6D6F6279002D2D6C6F67002F72756E2F636F6E7461696E6572642F696F2E636F6E7461696E6572642E72756E74696D652E76322E7461736B2F6D6F62792F34363636303563613733343038666536666363353065666234
[  511.755544] audit: type=1334 audit(1684929014.566:399): prog-id=130 op=UNLOAD
[  511.755575] audit: type=1334 audit(1684929014.566:400): prog-id=129 op=UNLOAD
[  511.756638] audit: type=1334 audit(1684929014.570:401): prog-id=131 op=LOAD
```

Elarion245, 2023-05-25:
Can this issue be reopened and the label be extended to 4C+? or should the 4C+ move into a separate topic?

Elarion245, 2023-05-25:
I understand that it is a kernel issue, so it is not about your port. Yet it may help others to see more quickly that there is something unstable. 

Using modprobe I get:
[ 3156.645542] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[ 3156.645973] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[ 3156.660966] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[ 3156.661055] brcmfmac mmc2:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.radxa,rock-4c-plus.bin failed with error -2
[ 3159.432332] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[ 3159.438890] ieee80211 phy0: brcmf_bus_started: failed: -110
[ 3159.445174] ieee80211 phy0: brcmf_attach: dongle is not responding: err=-110
[ 3159.470572] brcmfmac: brcmf_sdio_firmware_callback: brcmf_attach failed

What's also strange is that the Bluetooth does not seem to be working anymore either. Maybe I was lucky yesterday morning and it worked for a single attempt? Since then I haven't received any update over BLE. It sounded like the WIFI and BLE issues may both be related to the kernel?

Was the complete dmesg of the Rock 4C+ (without SSD option - no pi) of any general help?

citruz, 2023-05-25:
Yeah Bluetooth and Wifi are the same chip. If it cannot be initialized correctly, both will likely not work.

> Was the complete dmesg of the Rock 4C+ (without SSD option - no pi) of any general help?

It was helpful to confirm that there are no other errors that we missed ;)

Also the modprobe log confirms to me that this timeout issue is the root of the problem. For reference, this it what it looks like when I do the same:
```
[135305.544656] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43456-sdio for chip BCM4345/9
[135305.545052] brcmfmac mmc2:0001:1: Direct firmware load for brcm/brcmfmac43456-sdio.radxa,rockpi4b-plus.bin failed with error -2
[135305.646150] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/9 wl0: May 14 2020 17:26:08 version 7.84.17.1 (r871554) FWID 01-3d9e1d87
```

After some further digging, I found [this patch](https://lore.kernel.org/linux-arm-kernel/20180930150927.12076-1-hdegoede@redhat.com/) which also contains the exact same error messages. It seems like these can be the result of many different problems. As a last attempt I could try to build an image with this workaround to see if it helps.

Elarion245, 2023-05-25:
If it is not too much of an issue, I would for sure like to test it :)

Concerning no other errors: I think there is something about about not bring initialized properly in the dmesg, but I guess this is not really used by anyone anyhow...

citruz, 2023-05-26:
Please try with this image: https://github.com/citruz/haos-rockpi/releases/download/10.1%2B20230523beta/haos_rock-4c-plus-10.1.dev20230525.WIFI.FIX.img.xz

Elarion245, 2023-05-26:
You are quick! I am not at home atm, but will try Sunday or Monday and report back. Thanks for your work and effort!

Elarion245, 2023-05-28:
The fix works perfectly!

Complete dmesg
```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.1.29 (builder@83fc2b1d319e) (aarch64-buildroot-linux-gnu-gcc.br_real (Buildroot -gc00352ed6) 11.3.0, GNU ld (GNU Binutils) 2.38) #1 SMP PREEMPT Thu May 25 22:11:28 UTC 2023
[    0.000000] Machine model: Radxa ROCK 4C+
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000ff1a0000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] efi: UEFI not found.
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] NUMA: Faking a node at [mem 0x0000000000200000-0x00000000f7ffffff]
[    0.000000] NUMA: NODE_DATA [mem 0xf77d8380-0xf77dafff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x00000000f7ffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x00000000f7ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x00000000f7ffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] cma: Reserved 32 MiB at 0x00000000efe00000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 21 pages/cpu s45480 r8192 d32344 u86016
[    0.000000] pcpu-alloc: s45480 r8192 d32344 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 [0] 4 [0] 5
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Fallback order for Node 0: 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 999432
[    0.000000] Policy zone: DMA
[    0.000000] Kernel command line: zram.enabled=1 zram.num_devices=3 net.naming-scheme=v250 systemd.machine_id=fc71aca0099b4683a1fb8a60d39cc899 fsck.repair=yes  root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd rootfstype=squashfs ro rootwait rauc.slot=A earlycon=uart8250,mmio32,0xff1a0000 console=ttyS2,115200n8
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 3910168K/4061184K available (17920K kernel code, 3804K rwdata, 9336K rodata, 7808K init, 638K bss, 118248K reserved, 32768K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=6, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=6.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=6
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 256 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: GICv3 features: 16 PPIs
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000fef00000
[    0.000000] ITS [mem 0xfee20000-0xfee3ffff]
[    0.000000] ITS@0x00000000fee20000: allocated 65536 Devices @480000 (flat, esz 8, psz 64K, shr 0)
[    0.000000] ITS: using cache flushing for cmd queue
[    0.000000] GICv3: using LPI property table @0x0000000000440000
[    0.000000] GIC: using cache flushing for LPI property table
[    0.000000] GICv3: CPU0: using allocated LPI pending table @0x0000000000450000
[    0.000000] GICv3: GIC: PPI partition interrupt-partition-0[0] { /cpus/cpu@0[0] /cpus/cpu@1[1] /cpus/cpu@2[2] /cpus/cpu@3[3] }
[    0.000000] GICv3: GIC: PPI partition interrupt-partition-1[1] { /cpus/cpu@100[4] /cpus/cpu@101[5] }
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000001] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[... 769 lines cut ...]
[  257.851873] audit: type=1327 audit(1685285377.085:341): proctitle=2F7573722F7362696E2F69707461626C6573002D2D77616974002D74006E6174002D4400504F5354524F5554494E47002D7000746370002D73003137322E33302E33332E30002D64003137322E33302E33332E30002D2D64706F72740038383834002D6A004D415351554552414445
[  258.164287] hassio: port 6(veth2176b0d) entered disabled state
[  258.164966] veth8776c1f: renamed from eth0
[  258.254095] hassio: port 6(veth2176b0d) entered disabled state
[  258.254992] device veth2176b0d left promiscuous mode
[  258.255024] hassio: port 6(veth2176b0d) entered disabled state
[  259.362081] hassio: port 10(veth5508ec2) entered disabled state
[... cut ...]
[    7.034324] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    7.047071] brcmfmac mmc2:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.radxa,rock-4c-plus.bin failed with error -2
[    7.230838] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Nov  1 2021 00:37:25 version 7.45.241 (1a2f2fa CY) FWID 01-703fd60
```

Out of curiosity:
How did you include this fix? Did you compile the kernel manually with this fix on top? What does this mean for future releases of yours?

citruz, 2023-06-01:
> The fix works perfectly! Thanks so much!

Oh wow, great to hear that. I didn't really expect that to work. Did you notice any problems though? Any speed or stability issues?

> How did you include this fix? Did you compile the kernel manually with this fix on top?

I added a patch for the kernel which removes the three attributes from the 4C+'s device tree: https://github.com/citruz/haos-rockpi/commit/9e922408121959245cb375ca2605d45315bce945
That patch is automatically applied during the build process.

> What does this mean for future releases of yours?

The best way to include this is the future would probably be to create a device tree overlay. That way it can be enabled/disabled by every user individually if they experience problems. I could change the default config for the 4C+ so that the overlay is active by default.
I will need a few days for that though. HAOS 10.2 has just been released, so I will include the change with that new version.

Elarion245, 2023-06-01:
Sounds great! There are no stability issues. Wifi and Bluetooth are working flawlessly with the patch. The overlay sounds like a good option for the future.

citruz, 2023-06-03:
Alright. I just published a new beta based on 10.2 with the new overlay to disable the oob interrupts: https://github.com/citruz/haos-rockpi/releases/tag/10.2%2B20230601beta

Unfortunately, it is not possible to remove a device tree property as part of an overlay, you can only override it. Therefore I set the interrupts to an empty value. That results in a new error `brcmfmac: brcmf_of_probe: interrupt could not be mapped` which can be safely ignored. It's not ideal but this way the workaround can be easily disabled in case it causes problems.

I activated the overlay by default on all boards since I could see any negative impact either. That means you should be able to flash the new image and wifi should work out of the box. @Elarion245 

@borconi can you please test the 4B image? The wifi should now be working immediately after boot.

Elarion245, 2023-06-03:
I can confirm that wifi works out of the box with the new beta - well done!

The error you mention appears in dmesg as announced.

citruz, 2023-06-07:
Thanks for testing! I just published the same images as 10.2 (non-beta) and updated the README.
Glad that we were able to fix this :)

Elarion245, 2023-06-07:
Me too! Very glad you found this!
Your version is running very stable and shows great performance on the 4c+. Only remaining step for me is transiting to Emmc.