citruz/haos-rockpi #12: No CPU or GPU temperature readings

Elarion245, 2023-05-22:
### Describe the issue you are experiencing

After Boot, System Monitor finds no CPU Temperature or GPU Temperature sensors. Indeed /sys/class/thermal ist not showing any thermal_zone* devices.

More information is available here: https://forum.radxa.com/t/rockpi-4-cpu-temperature/9530
sounds like an overlay for tsadc activation is required.

### What operating system image do you use?

rock-4c-plus

### What version of Home Assistant Operating System is installed?

10

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Check /sys/class/thermal after boot

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
No particular message. Looks like a disabled feature
```

citruz, 2023-05-22:
Interesting, this only seems to apply to the 4C+. All other boards inherit from `rockchip/rk3399-rock-pi-4.dtsi` which includes the following snippet to activate tsadc:
```
&tsadc {
	status = "okay";

	/* tshut mode 0:CRU 1:GPIO */
	rockchip,hw-tshut-mode = <1>;
	/* tshut polarity 0:LOW 1:HIGH */
	rockchip,hw-tshut-polarity = <1>;
};
```
I can try to add that into the 4C+ device tree, but I would need you to test it.

Elarion245, 2023-05-23:
Sounds great! I will test and report back once you are comfortable with an update. Thank you!

Elarion245, 2023-05-24:
Don't know if you changed something on the new beta, but it had no impact on the missing temperature sensors on the 4C+

citruz, 2023-05-24:
No I didn’t change anything yet, but I am planning to post a second beta with this change in the next days!

citruz, 2023-05-24:
Please try https://github.com/citruz/haos-rockpi/releases/download/10.1%2B20230523beta/haos_rock-4c-plus-10.1.dev20230524.WITH.TSADC.img.xz

I patched the device tree directly, no need for an overlay.

Elarion245, 2023-05-25:
Works perfectly!

[image]

The systemmonitor now finds the CPU temperature. I was also able to manually read out the GPU-temp via the following sensor definition
```
- platform: command_line
    name: "GPU Temperature"
    command: "cat /sys/class/thermal/thermal_zone1/temp"
    unit_of_measurement: "°C"
    value_template: '{{ value | multiply(0.001) | round(1) }}'
    device_class: "temperature"
```

citruz, 2023-05-25:
Oh that's cool, thanks for posting the sensor configuration! Added those two to my dashboard as well.

Elarion245, 2023-05-25:
I'm glad, if there is something that I can provide that helps you ;)

citruz, 2023-05-28:
Implemented in latest release https://github.com/citruz/haos-rockpi/releases/tag/10.1%2B20230526

closed 2023-05-28 by citruz