citruz/haos-rockpi #13: Missing possibility to disable LEDs

Elarion245, 2023-05-22:
### Describe the issue you are experiencing

The green power and blue status Led cannot be disabled easily. There should be flags in the haos-config.txt defining if they shall be active or not.

The only workaround I was able to find is described here: https://community.home-assistant.io/t/disable-leds-on-rock-pi/573294/2

Still, a configuration entry would be preferable to a workaround breaking the container... 

### What operating system image do you use?

rock-4c-plus

### What version of Home Assistant Operating System is installed?

10

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

N/A

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
See attached link for a workaround on how to disable the leds. Some guys may also wish to disable the network Leds. Not sure what the commands for that would need to look like.
```

citruz, 2023-05-22:
I am working on adding support for devicetree overlays in the next release. Using these it will be possible to disable the LEDs in the haos-config.txt.
I already have a working overlay to disable the heartbeat led on my 4B+. Unfortunately, the green power led is soldered directly to the voltage supply, so it cannot be turned off on any board except the 4C+ (see https://forum.armbian.com/topic/10980-rockpi4-510-rockchip64-leds/?do=findComment&comment=82886).
For the 4C+ it should work, but I will need to rely on you for testing. I will let you know once I have an image to test.

Elarion245, 2023-05-23:
Sounds great! Will definitely test this, once you are ready

citruz, 2023-05-24:
I just published a beta release which adds DT overlays for disabling the LEDs (separately or both).
https://github.com/citruz/haos-rockpi/releases/tag/10.1%2B20230523beta
Just edit `haos-config.txt` on the boot partition, uncomment the example `overlays=` line and reboot.
Let me know if you run into any problems.

Elarion245, 2023-05-24:
I just flashed the image - disabling the LEDs via the overlay= works perfectly! Thank a LOT! Now I can go ahead and remove the workaround solution which breaks the Docker container. 
(I haven't tested only disabling one, but I guess if disabling both works fine, then also disabling them individually should be fine)

citruz, 2023-05-28:
Implemented in latest release https://github.com/citruz/haos-rockpi/releases/tag/10.1%2B20230526

closed 2023-05-28 by citruz