citruz/haos-rockpi #14: Boot from Nvme Issue

gavinjai, 2023-05-24:
### Describe the issue you are experiencing

I have a RockPi 4 C+.  I flashed haos image on the NVMe in Windows first using balenaEtcher, then use the same software to flash miniloader.img on the microSD card after i clean the partition using diskpart in Windows.  

But Upon boot It is not recognizing the nvme and is trying to load the boot partition but failed.  Is there a reason why I need to flash the haos image with armbian installed on SD card first? 
[image]

### What operating system image do you use?

rock-4c-plus

### What version of Home Assistant Operating System is installed?

10

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. reflash HAOS on SSD
2. reflash miniloader
3. boot without any USB device connected
...

### Anything in the Supervisor logs that might be useful for us?

```txt
n/a
```

### Anything in the Host logs that might be useful for us?

```txt
n/a
```

citruz, 2023-05-25:
>  Is there a reason why I need to flash the haos image with armbian installed on SD card first?

No, if you are able to connect the SSD to your PC and flash it from there, that is totally fine.

> But Upon boot It is not recognizing the nvme and is trying to load the boot partition but failed

It looks like you are right about u-boot not detecting the nvme. This line seems to indicate some problem with initializing the PCIe bus:  `rockchip_pcie pcie@f8000000: failed to get vpcie1v8 supply (ret=-121)`

A few ideas:
1. In the [radxa wiki ](https://wiki.radxa.com/Rockpi4/FAQs#Which_M.2_SSD_are_supported.3F)there is a list of requirements for SSDs to work with known working and not-working models. Does yours satisfy the requirements?
2. You could verify that the SSD works in general on your board by flashing armbian to the SD card, boot it and check whether the SSD is recognized.
3. Do you have a serial adapter? That would allow us to do some further debugging in u-boot.
4. Please also try the latest beta release which contains a newer version of u-boot.

gavinjai, 2023-05-25:
Hi,

1.  I am using Kioxia 256GB M.2 2230 NVMe Drive.  Drive was detected when Armbian and Debian is installed. I followed this guide https://jamesachambers.com/radxa-rock-pi-4c-plus-ssd-boot-guide/ and able to boot the NVME with a dummy SD card.  But i wanted to just have HAOS on the nvme.
2. See above
3. Unfortunately I don't. 
4. I tried the beta miniloader release and got a different msg (ignore the USB msg since I plugged in the USB hub with the keyboard and mouse). I did not flashed the beta HAOS image yet on my SSD. I figured I don't need unless the miniloader has to work in together with the same HAOS release.
[image]

Elarion245, 2023-05-25:
Why does it say 4B+ on top when you do have a RockPi 4C+?

citruz, 2023-05-25:
The 4C+ image currently uses the `rock-pi-4-rk3399` defconfig for u-boot instead of  `rock-pi-4c-rk3399` which I should probably change. 
@gavinjai  can you try the Rock 4C (without the plus) image? That should display "ROCK Pi 4C" instead of B and mayyybe fixes the NVMe issue.

gavinjai, 2023-05-26:
I tried the image above and is still getting error.   Should I try it with a different HAOS image?

[image]

citruz, 2023-05-26:
Too bad. I'm sorry but I am running out of ideas. It's very hard to debug these kinds of issues without access to the hardware.
You could try asking in the radxa forum or discord if somebody has seen this error before.

citruz, 2024-05-02:
Closing due to inactivity

closed 2024-05-02 by citruz