citruz/haos-rockpi #17: Network storage unavailable

JoeMoer222, 2023-06-09:
### Describe the issue you are experiencing

Getting the following error when trying to connect network storage
The operating system does not support network storage
To use network storage you need to run at least Home Assistant Operating System 10.2

### What operating system image do you use?

rock-4se

### What version of Home Assistant Operating System is installed?

Operating System 10.2.dev20230602

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
no
```

### Anything in the Host logs that might be useful for us?

```txt
no
```

### System information

## System Information

version | core-2023.6.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.3
os_name | Linux
os_version | 6.1.29
arch | aarch64
timezone | Africa/Johannesburg
config_dir | /config

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 10.2.dev20230602
-- | --
update_channel | stable
supervisor_version | supervisor-2023.06.1
agent_version | 1.5.1
docker_version | 23.0.6
disk_total | 234.0 GB
disk_used | 4.0 GB
healthy | true
supported | true
board | rock-4se
supervisor_api | ok
version_api | ok
installed_addons | 

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
mode | auto-gen

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | June 9, 2023 at 9:54 AM
-- | --
current_recorder_run | June 9, 2023 at 12:03 PM
estimated_db_size | 1.30 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

citruz, 2023-06-09:
What are you trying to do? Which commands are you executing exactly to get to that error message?

JoeMoer222, 2023-06-09:
Settings/system/storage to add network storage.

citruz, 2023-06-12:
I was able to reproduce the problem and trace it back to this check in the supervisor:
https://github.com/home-assistant/supervisor/blob/72db559adcaee3435515786eacee74624a0875a1/supervisor/host/manager.py#L120
```
self.sys_os.version >= AwesomeVersion("10.2")
```

The check fails because this fork uses the `.devXXXX` suffix for the OS version:
```
>>> AwesomeVersion("10.2.dev20230602") >= AwesomeVersion("10.2")
False
>>> AwesomeVersion("10.3.dev20230602") >= AwesomeVersion("10.2")
True
```

So unfortunately the fix is to wait for the next release.

JoeMoer222, 2023-06-12:
I suspected that that could be the problem. Will wait for the fix with baited breath. Thanks and keep up the good work.

Elarion245, 2023-06-25:
Any news on the 10.3 release?

citruz, 2023-06-25:
It has already been built but not tested yet as I am on vacation.
But if you feel adventurous: https://github.com/citruz/haos-rockpi/actions/runs/5350123234
(Realistically everything should work)

Elarion245, 2023-06-25:
No rush. Enjoy your vacation 😉

Elarion245, 2023-06-29:
I can confirm that it works. Also the network storage works. Only editorial thing is that in the haos-config the PWM overlay is not mentioned as an option. Great work!

citruz, 2023-06-30:
10.3 has just been released: https://github.com/citruz/haos-rockpi/releases/tag/10.3%2B20230622

closed 2023-06-30 by citruz