citruz/haos-rockpi #18: Features Request: Fan PWM control

Elarion245, 2023-06-11:
### Describe the issue you are experiencing

It would be nice to enable PWM fan control via an overlay.

It seems that it is possible: https://forum.armbian.com/topic/15341-rock-pi4-pwm-control-no-overlay/

I am currently running on passive cooling with heat sinks, but I do have a fan at hand. So far I am not using it, cause I do not want it to make noise all the time. During daytime it would be fine though.
I hope to achieve this via PWM and an home assistant automation.

Thank you for taking a look! 

### What operating system image do you use?

rock-4c-plus

### What version of Home Assistant Operating System is installed?

10.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

N/A

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
N/A
```

citruz, 2023-06-12:
Creating a device tree overlay which activates the pwm pins should be straight forward (honestly, the one from the [repo](https://github.com/PetrozPL/rockpi4_pwm_overlay) could even work out of the box if you want to try it). But then you need some way to set the value. Using a script which just writes to the device directly (e.g. `echo 500000000 > /sys/class/pwm/pwmchip0/pwm0/duty_cycle`) _could_ work, but I am not sure.

Elarion245, 2023-06-12:
Sounds great! I am not sure either if it is supposed to work without breaking the container, however I can get root access using e.g. the Advanced SSH & Terminal Add-On or https://community.home-assistant.io/t/run-on-startup-d/271008.

I do not have the tool-chain to compile an overlay. Would you mind releasing a dev-Version? I could then also try my new EMMC update strategy via xRDP and DietPi ;)

citruz, 2023-06-13:
[rk3399-pwm-gpio.dtbo.zip](https://github.com/citruz/haos-rockpi/files/11739432/rk3399-pwm-gpio.dtbo.zip)
Here you do. Just put that into the overlays folder on the boot partition and adjust `haos-config.txt`. Tested it on my board and the pwm devices did indeed show up.
However, I hooked an LED up to one of the pins but wasn't able to get it to work. But maybe I did something wrong.

Btw, for real SSH access to the host you want this plugin: https://community.home-assistant.io/t/add-on-hassos-ssh-port-22222-configurator/264109
Follow the instructions and you will be able to ssh into the host.
The SSH & Terminal Add-On only gives you access to a restricted container, not the real host device.

Elarion245, 2023-06-15:
Tried it myself now as well. Good and (for me) bad news..

Good:
PWM works. PINs 11 and/or 13 change according to the value i set. Essentially I am using the pwmset script from the repo with the slight change of changing pwmchip1 and pwmchip2 to pwmchip0 and pwmchip1.

@citruz: did you send `echo 0 > /sys/class/pwm/pwmchip${_cidx}/export` before trying the echo on the duty-cycle?

Now on the bad news:
I also got one of the 2-Pin fans with my Rock 4C+. I was not overthinking it and simply assumed that this type of model would not require a dedicated speed line but would simply be controllable out of the box where the 5V pin would be PWM'ed as well. Should have given it a bit more thinking to begin with.

Anyhow: I now have three scripts: pwmset, read and activate. Now I only need to buy a new fan (three wire, proper dimensions) to realize what I had in mind initially.

Thank you for your effort! You should include the overlay in future releases. It is working fine.

A short side-node on root access:
I used a direct SSH on 22222 for my tests. The Repo talks about a digital user group, which is not existing in HAOS. I am not sure if there is any simple means in HAOS to allow modification of the duty-cycle from the usual homeassistant front-end.

On the add-on for root: Your statement for the other plugin is only partially true. The add-on you mention works, however the following works as well: 
1) Disable protected mode of the Advanced SSH & Web Terminal
2) Launch the Terminal - no real root yet
3) ssh 127.0.0.1 -p22222 -i /config/misc/private_key

As a proof of concept (I've modified the read script to take the PWM pin as argument):
[image]

Sending this command should be possible from within home-assistant, allowing some crude control over the fan-speed via automations + creating a template-sensor for the current fan-speed.

citruz, 2023-06-15:
> @citruz: did you send echo 0 > /sys/class/pwm/pwmchip${_cidx}/export before trying the echo on the duty-cycle?

I did. I was able to access the `pwmchip0/pwn0` device and modify the variables. Tried the script and setting it manually. I connected the PWM pin to the LED anode and the cathode to ground with a resistor. That should work right? (Connecting the LED just to 5V and ground worked, so the LED and resistor are working)

> Thank you for your effort! You should include the overlay in future releases. It is working fine.

Great, I will. HAOS 10.3 beta was released today so as soon as that is released, I will include the pwm overlay with the next version of haos-rockpi.

> ssh 127.0.0.1 -p22222 -i /config/misc/private_key

Sure, but that requires that you set up the "real" ssh beforehand doesn't it?

> Sending this command should be possible from within home-assistant, allowing some crude control over the fan-speed via automations + creating a template-sensor for the current fan-speed.

I think it will be even easier than that. Because the devices are exposed as writable in the main home assistant container, you can actually just run a script from HA:

```
shell_command:
  enable_pwm: echo 0 > /sys/class/pwm/pwmchip0/export
```

Elarion245, 2023-06-16:
That with the led should have worked flawlessly. I simply used a voltmeter to measure the voltages between pins 11,13 and GND, which behaved according to the commanded percentage.

For ssh on 22222 I used putty to generate a keypair and then put the public key on the rock and used ha is import.
then private key as well on it (not very secure xD) for above command. By the way: above command does not work with the terminal in the vs code add on. Only the advanced terminal and ssh add on allows me to disable private mode, which is when the command succeeds.

Thanks for your comment about the exposure of the devices. This makes it even easier then. But first I need to do some soldering or find a new fan...

Elarion245, 2023-06-17:
Does HA device exposure only work in the context of a shell command? I've tried the following without success:
[image]

With root it works...

Elarion245, 2023-06-18:
Okay, so I finally got it working. There may be more elegant ways to do this (please share yours), but below fulfills its purpose:

1) To automatically enable the PWM upon boot, I use the following add-on: https://community.home-assistant.io/t/run-on-startup-d/271008
In my case, this calls /config/pwm/pwmenable 0, which is just a self-written script to write the export 0 to pwm 0 or 1.
2) In my configuration.yaml I added the following sensors:
```
shell_command:
  set_pwm_off: 'ssh 127.0.0.1 -p22222 -i /config/misc/private_key -o StrictHostKeyChecking=no "/mnt/data/supervisor/homeassistant/pwm/pwmset 0 0"'
  set_pwm_min: 'ssh 127.0.0.1 -p22222 -i /config/misc/private_key -o StrictHostKeyChecking=no "/mnt/data/supervisor/homeassistant/pwm/pwmset 0 28"'
  set_pwm_mean: 'ssh 127.0.0.1 -p22222 -i /config/misc/private_key -o StrictHostKeyChecking=no "/mnt/data/supervisor/homeassistant/pwm/pwmset 0 50"'
  set_pwm_max: 'ssh 127.0.0.1 -p22222 -i /config/misc/private_key -o StrictHostKeyChecking=no "/mnt/data/supervisor/homeassistant/pwm/pwmset 0 100"'
```
you obviously can alter the location of your private key as you wish. Originally, these commands resulted in a 255 error.  I then found this entry: https://community.home-assistant.io/t/remote-ssh-command-error-255/187891/7 which stated that I had to copy /root/.ssh to /config/.ssh 
I was able to do the copy from a regular terminal. via cp -r /root/.ssh /config/.ssh
This copies the public key to the place required for automation of this ssh call later.

3) Next, I added a sensor for the fan-speed:
```
    #Add sensor for current fan speed in %
  - sensor:
      name: "FAN speed"
      unique_id: "fan_speed_percent"
      command: 'ssh 127.0.0.1 -p22222 -i /config/misc/private_key -o StrictHostKeyChecking=no "/mnt/data/supervisor/homeassistant/pwm/pwmread 0"'
      unit_of_measurement: "%"
      value_template: "{{ value }}"
```
pwmread is the pwmsts from the above repo where I modified the argument to the pwm interface used

4) Next, I created the following scripts: Fan Off, Fan low, Fan middle and Fan Max, which are simply linking to the shell-commands. This step is not really required, as you may also execute the shell-commands from an automation directly, however it was handy during testing, so I kept them in and used them in the automations

5) Finally, I defined the following automations:
Fan off, low, middle, max, where each is similar to the following with different temperature boundaries
```
alias: Fan low
description: ""
trigger:
  - platform: time_pattern
    minutes: /1
condition:
  - condition: numeric_state
    entity_id: sensor.processor_temperature
    above: 67
    below: 75
action:
  - service: script.fan_min
    data: {}
mode: single
```

Put this on a nice dashboard to get (I added a line to mark the point in time where the setup was completed):
[image]

I am using 67°C to set the fan to 28% (min for me to get it constantly turning), then 50% between 67 and 75°C and finally 100% above 79°C. In the Fan-low setting I can only hear the fan when I am directly next to the RockPi, so this is very fine. Starting at Fan middle it is noticable when just walking past. Since I am aiming for passive cooling where active only takes place when really required, I set the boundary to 67° for the time being.

Using ssh on its own the root access in the commands is not optimal, but it works. Direct exposure of the pwm devide did not work for me (returned error code 1, which apparently is not very specific).

Thanks again for your help in enabling this feature! Very nice setup now running smooth on the Rock 4C+!

citruz, 2023-06-30:
10.3 has just been released which contains the PWM overlay: https://github.com/citruz/haos-rockpi/releases/tag/10.3%2B20230622

closed 2023-06-30 by citruz