citruz/haos-rockpi #20: Rock 4 SE - Boot from NVME fail without display connected

andcompe, 2023-08-10:
### Describe the issue you are experiencing

Hi

I've got a Rock 4 SE and a Kingston NVME SSD. HAOS boots fine as long as I keep the display connected, but as soon as I remove the HDMI connector before power on, it fails. What really bothers me is that it is the NVME that fail after the kernel is loaded. Anyway, please see attached logs from booting with and without HDMI connected. 

I have tested for several hours with different settings in haos-config.txt, but when I change that file to anything else than I hade during install, it will not boot at all. haos-config.txt is the original from the image, except I added the overlay to turn off the blue led.

Otherwise, I really like this image. It is nice to run HA OS even on my Rock. It is supposed to replace my old RPi 3b.

[boot with HDMI connected.txt](https://github.com/citruz/haos-rockpi/files/12316248/boot.with.HDMI.connected.txt)
[boot without HDMI connected.txt](https://github.com/citruz/haos-rockpi/files/12316249/boot.without.HDMI.connected.txt)

### What operating system image do you use?

rock-4se

### What version of Home Assistant Operating System is installed?

6.1.39

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

Don't know if it is my device, but this is what I did:

1. Installed Ubuntu (Radxas official version) on SD and booted.
2. From Ubuntu, installed HAOS to NVME.
3. Created a miniloader SD using balena in Windows.
4. Booted from SD miniloader. HAOS boots fine. 
5. Configured HAOS. Works nice.
6. Removed HDMI. Boot failed.
7. Attached HDMI to see why. Boot successful.
8. Removed HDMI. Boot failed.
9. Attached serial. Boot failed. See logfile.
10. Attached HDMI. Boot successful. See logfile.

Of course there is quite a few trial and errors in between.

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
Attached serial logs. See nvme at the end.
```

citruz, 2023-08-11:
Huh. Very weird issue indeed. This will be hard for me to debug but here are some ideas you could try:
- Check if the Radxa provided image is able to boot from NVMe without HDMI connected (to check whether the issue is specific to this image or to the hardware)
- Check if there are firmware upgrades for your SSD: https://www.kingston.com/en/support/technical/ksm-firmware-update
- Try with an older version of the image, especially one with a 5.x kernel: https://github.com/citruz/haos-rockpi/releases/tag/9.5%2B20230204 (there is no 4SE specific image but should work nevertheless)

andcompe, 2023-08-13:
Thank you for your reply. 

I have now tested a lot of things. Debian loaded fine from NVME without the screen. Firmware was for Windows only, so I could not do that.

Anyway, it is working now. I completely wiped the SD card and flashed the miniloader image again, but this time I used Debian (dd) and not balena. Now it starts fine without the display connected. I would like to know why it booted at all if the miniloader was not correct before, but I guess I have to stick with the fact that it works now.

citruz, 2023-08-15:
Well.. I am as curious as you. But glad that it works now!

closed 2023-08-15 by citruz