citruz/haos-rockpi #21: Support for rockpro64

patkino, 2023-08-14:
### Describe the issue you are experiencing

I have a chinese rk3399 tvbox, similar to rockpro64 board (libreelec with rockpro64 dtb runs flawlessly). Is it possible to adapt this build to run on this type of hardware? Tried all rock-pi images with rockpro64 dtb's - no luck, doesn't boot.
Would be thankful for any advice.

### What operating system image do you use?

rock-4c-plus

### What version of Home Assistant Operating System is installed?

6.6

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Boot haos on rockpro64 

### Anything in the Supervisor logs that might be useful for us?

```txt
.
```

### Anything in the Host logs that might be useful for us?

```txt
.
```

### System information

.

citruz, 2023-08-23:
Yes that should be possible, both Linux and u-boot seem to have configs for the rockpro64. I may be able to provide you with an image to test next week, but no guarantees.

citruz, 2023-08-26:
Please test this image: https://github.com/citruz/haos-rockpi/releases/tag/10.5%2B20230826beta

citruz, 2023-09-10:
Any updates @patkino?