citruz/haos-rockpi #22: Preparing Home Assistant - not completing

DuncanLHS, 2023-10-21:
### Describe the issue you are experiencing

Not getting past the "Preparing Home Assistant" Screen.

### What operating system image do you use?

rockpi-4b-plus

### What version of Home Assistant Operating System is installed?

10.5

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

Write image to emmc on-board storage, boot.

### Anything in the Supervisor logs that might be useful for us?

```txt
As per https://github.com/home-assistant/supervisor/issues/4564#issue-1899851866
```

### Anything in the Host logs that might be useful for us?

```txt
I don't think so?
```

### Additional information

I've had a couple of issues getting this up and running. On first boot the web interface wasn't available at all. FING reported that the port wasn't open. A power cycle fixed that.

I'm now getting stuck at the "Preparing home assistant" screen. I've tried to dig into logs but am also being affected by the issue linked above. Not sure if that's the root cause but it's making it difficult to find anything meaningful in logs.

Connectivity appears to be ok (wired connection), DNS address is correct. New to Home Assistant (and not all that experienced with linux) and I'm not sure where to go next to troubleshoot. Happy to dig further with some pointers.

Elarion245, 2023-10-22:
If you are still connected via hdmi, you can read the logs (ha su logs or ha core logs) depending on the component. This should give you more details. 
Also you can push the blue pulsating ball to see what is going on. I am waiting for the release of 11.0, but 10.4 works flawless on Emmc in my 4c+.

DuncanLHS, 2023-10-22:
OK thanks!

In the su logs I've got
`ERROR (MainThread) [supervisor.docker.interface] Can't install ghcr.io/home-assistant/qemuarm-64-homeassistant:2023.10.3: 404 Client Error for http+docker://localhost/v1.42/images/ghcr.io/homeassistant/gemuarm-64-homeassistant:2023.10.3/json: Not Found ("No such image: ghcr.io/home-assistant/qemuarm-64-homeassistant:2023.10.3")`
_*Yes, I typed that out from the HDMI :)_

Under the blue dot output the only issue I can find is:
`WARNING (MainThread) [supervisor.updater] Board 'rockpi-4b-plus' not found`
Which i'm guessing is expected when checking for updates as it's not an officially supported board?

Elarion245, 2023-10-22:
The warning can be ignored, but supervisor reports that it cannot find the image of home assistant core (the part you are wanting to interact with).

You can now Google the root cause of the 404 (supervisor not being able to fetch the core image). Sounds to me like an internet access issue. Are you connected over ethernet?

citruz, 2023-10-24:
Yeah that image should definitely exist: https://github.com/home-assistant/core/pkgs/container/qemuarm-64-homeassistant/137169558?tag=2023.10.3

I remember an issue where the github docker repositories were down temporarily but this doesn't look like it. Please do all the usual internet connection troubleshooting as @Elarion245 suggested. 
If you are in the homeassistant prompt, you can enter `login` to get into the real shell. `curl` should be installed in the image.

Elarion245, 2023-10-24:
@DuncanLHS I don't know how far your installation went, but if your connection works you can try updating the core via ha core update with an optional - - version tag to check if any other image results in the same error for you (we are now at 2023.10.5)

@citruz: any guess on when you will find the time to release 11.0?

citruz, 2023-10-24:
@Elarion245  I am currently finishing up with testing. If all goes well, it will be released tomorrow.

DuncanLHS, 2023-11-01:
So I tried 11.0 (just in case) but same issue again, it's a networking problem. Odd, because I haven't got a complex setup by any means. It's not a hardware issue because Debian is fine so I'm a bit lost but will no doubt have another go. Perhaps plugging directly into my router is worth a pop just to get it setup. It might be a little easier to diagnose any issues once the OS is fully up and running.

Thanks for the pointers but clearly not a problem with the image here so by all means close the issue, other ise I'll come back and report if and what I find when I manage to sort it out.

citruz, 2023-11-01:
We can try to help if you describe your setup a little bit further. How are you connected to your router?
What kind of networking troubleshooting have you already done and what were the results?

Elarion245, 2023-11-02:
You should really complete the initial setup on Lan and once you can login to home assistant configure the wlan

closed 2023-11-17 by citruz

DuncanLHS, 2024-01-30:
Update, after some weirdness with other devices round the house I discovered a configuration issue in my router. A DNS issue, IT'S ALWAYS DNS!!! Most addresses were resolving fine, others not. After fixing that I'm now happily up and running with HAOS on my RockPI! Thanks for the patience and pointers.

citruz, 2024-01-30:
Glad you got it working. It's always DNS or IPv6.