citruz/haos-rockpi #23: Nanopi M4V2 support

NachoFdV, 2023-12-02:
### Describe the issue you are experiencing

I am trying to get a HAOS image working for the Nanopi M4V2. So far none of the images boot (even the RockPro64).
I am using the program disks (gnome-disk-utility & UDisks) to flash the images but only DietPi and ArmBian images worked. 

### What operating system image do you use?

rockpi-4a

### What version of Home Assistant Operating System is installed?

none

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

Burn the RockPi image to the SD card an boo the Nanopi M4V2 without the emmc module

### Anything in the Supervisor logs that might be useful for us?

```txt
no
```

### Anything in the Host logs that might be useful for us?

```txt
no
```

citruz, 2023-12-09:
Let me know if this works: https://github.com/citruz/haos-rockpi/releases/tag/11.2%2B20231209

NachoFdV, 2023-12-09:
Thank you Citruz!
Unfortunately I copied to the sd-card using the xzcat & dd commands and doesn't' boot.
I tryed in the sd-card and in the emmc but so far no luck.
Only DietPi and ArmBian works. And https://sd-card-images.johang.se/boards/nanopi_m4.html if I have ArmBian in the emmc.

citruz, 2023-12-10:
When you tried booting from sd card, did you have another image in the emmc? If yes, it would be a good idea to fully wipe the emmc before trying to boot from sd so that it is forced to use the bootloader from the sd.
The only feasible option to debug this is with serial output so that we can see what is going wrong. Do you have a USB-serial adapter?
Using a different kind of memory (ddr3 vs ddr4) sounds like a pretty good reason why the M4 image wouldn't work on an M4V2. I saw some patches being sent around for the M4V2 which could be worth a try but without the ability to debug over serial it's not ideal.

NachoFdV, 2023-12-10:
I don't have the USB-serial  and I removed the emmc before power on the NanoPi with the sd.
Could be that the memory make the different but both sbc have the same link to download the official images from FriendlyElec.
Thanks for your help I will keep playing and if I success I will let you know.

citruz, 2023-12-12:
I updated the release with an m4v2 image. Let me know if that works.

NachoFdV, 2023-12-14:
So far no luck. 
Thanks for your work!!

citruz, 2023-12-14:
In that case serial logs are the only way to go forward. These adapters are cheap ;)