citruz/haos-rockpi #24: boot fails on R4SE

MrZajda, 2024-01-03:
### Describe the issue you are experiencing

HW :Rock 4SE SDcard installation
after power on loader reports : 
Model: Radxa ROCK Pi 4A         _why?

..ends with message "Starting kernel" and freeze afterwards.

### What operating system image do you use?

rock-4se

### What version of Home Assistant Operating System is installed?

trying latest 11.2

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. install haos_rock-4se_11.2... on SD card using balena etcher
2.put SD card into rock 4SE sd slot
3. power on
...

### Anything in the Supervisor logs that might be useful for us?

```txt
not able to report
```

### Anything in the Host logs that might be useful for us?

```txt
it is not able to start with default setup
```

citruz, 2024-01-03:
> HW :Rock 4SE SDcard installation after power on loader reports : Model: Radxa ROCK Pi 4A _why?  

Some boards have no specific configuration for the uboot bootloader so the generic config is used which shows up as 4A. That has worked in the past and shouldn’t be an issue though.
Can you please try older versions such as 9.5 or 10.0?

MrZajda, 2024-01-05:
Thanks for such quick response.
I have tried to boot with 10.0 and 9.5. as well with very same results - freezes after kernel starting.

Some screenshots:
[image]
ached.
[image]
[image]
and finally  with flooded display effect :)
[image]

What do you think might be the issue?

Petr

citruz, 2024-01-09:
The stripes on the display are, believe it or not, a good sign. It means that the kernel has started but probably crashed early in the initialization.
Unfortunately without serial logs this is impossible to debug. Do you have a USB to serial adapter?

MrZajda, 2024-02-05:
Dont have anything like that.. I am just beginner.

citruz, 2024-02-05:
Hmm.. the logs/screenshots you provided all look good. What happens after the stripes appear on the display? Do they stay?

MrZajda, 2024-02-06:
Hi ,

I dont think so...  it stays for a while and my monitor says "no hdmi
signal detected" after 10seconds (approximatelly).

BR

Petr

trucktire, 2024-05-06:
I tried running this and it reports crc error on my emmc.

DuncanDotPY, 2024-05-26:
Just to add, I've now run into the same issue on a 4c+. I had a long-running & stable homeassistant setup running on it that stopped working all of a sudden.

Initially, I tried to boot an old instance from another SD card which ran into the same issue. I reimaged that old SD with debian and it worked fine. Then reimaged again with different releases of haos-rockpi and all of them failed each time.

So I've ruled out that the SD cards are the issue (as I tried multiple and managed to boot debian). It shouldn't be a software issue considering it was working once upon a time and I tried different versions/older images? But also considering the same board worked for debian it doesn't overtly look like a hardware issue?

So I'm stumped - but I'm a bit of a noob at this stuff also. Any suggestions? I'm going to order another 4c+ and see will it boot my old HA instance. I'll order a USB to serial adapter too.

cafebeef, 2024-11-18:
This seems like an old post, but I tried it with Rock 4se and the latest 13.2.

The problem of stripes appearing on the HDMI display is the same.
However, if I left it as it was, a login prompt appeared on the serial console.
The Home Assistant CLI also appeared on the HDMI display.

I tried it without connecting a LAN cable, but it seems that the NTP time synchronization timeout at startup takes a long time.

I've attached the serial console log.
I hope this helps someone.

[serial_output.log](https://github.com/user-attachments/files/17797680/serial_output.log)