citruz/haos-rockpi #25: WIFI no longer works on 4c+ with version 11.4

Elarion245, 2024-01-28:
### Describe the issue you are experiencing

I've just upgraded from 11.0 to 11.4. Everything went smooth apart from the fact that the wifi interrupt overlay no longer works (wifi device not visible in HomeAssistant.

dmesg with respect to crcmfmac:
```
[    6.321634] brcmfmac: brcmf_of_probe: interrupt could not be mapped
[    6.391780] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    6.443179] brcmfmac mmc2:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.radxa,rock-4c-plus.bin failed with error -2
[    6.466300] usbcore: registered new interface driver brcmfmac
[    6.493146] brcmfmac mmc2:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.txt failed with error -2
[    7.120457] Bluetooth: hci0: BCM4345C0 'brcm/BCM4345C0.hcd' Patch
[    7.518450] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
```

Let me know if the full dmesg would be of help. Could it be that the interrupt mapping has changed and the overlay needs an update?

I have not tried any version between 11.1 and 11.3 (direct jump from 11.0 to 11.4). With 11. 0 the WIFI was still working fine.

### What operating system image do you use?

rock-4c-plus

### What version of Home Assistant Operating System is installed?

11.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Upgrade from 11.0 to 11.4 and boot HAOS

### Anything in the Supervisor logs that might be useful for us?

```txt
dmesg seems most helpful
```

### Anything in the Host logs that might be useful for us?

```txt
dmesg seems most helpful
```

citruz, 2024-01-28:
Is there an error in the uboot output that the overlay cannot be applied?

Elarion245, 2024-01-28:
How can I access the logs you have in mind?

citruz, 2024-01-28:
Watching the HDMI output or serial logs :)

Elarion245, 2024-01-28:
Damn. To check this, I need to grab some stuff from the cellar. No can do quickly, but I'll add it to my list. 

Comparing above with https://github.com/citruz/haos-rockpi/issues/11#issuecomment-1566164139 I see the following additional lines:
```
[    6.466300] usbcore: registered new interface driver brcmfmac
[    6.493146] brcmfmac mmc2:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.txt failed with error -2
[    7.518450] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
```
Why does it say firmware load for a text file?

citruz, 2024-01-28:
I just checked my rock pi and the wifi interface is not detected either. Damnit. I used to test wifi on every release but this was the first time I skipped it...
I will pull the release for now until I have more time to debug.

Elarion245, 2024-01-28:
Sorry to hear that. But at least it means two things:
1) 11.3 had it working as you checked it
2) You can debug on your system

citruz, 2024-01-31:
I accidentally enabled a config value with conflicted with the firmware package required to run the wifi chipset. You would think that this triggers some kind of error or at least warning.. but no.
Fixed and updated the release.

closed 2024-01-31 by citruz