citruz/haos-rockpi #27: Bluetooth not working

mistaoh, 2024-04-06:
### Describe the issue you are experiencing

Fresh install of 12.1 averything seems to be ok except bluetooth. No adapters found.

### What operating system image do you use?

rock-4c-plus

### What version of Home Assistant Operating System is installed?

Home Assistant OS 12.1.dev20240324

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Flash rockpi HA-OS image to SD
2. Start and restore HA backup from old raspberry pi3
3.  
...

### Anything in the Supervisor logs that might be useful for us?

```txt
2024-04-06 07:11:25.449 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state running
2024-04-06 07:11:25.959 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
2024-04-06 07:11:25.960 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state running
2024-04-06 07:11:25.960 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
2024-04-06 07:15:08.774 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
2024-04-06 07:15:08.896 WARNING (MainThread) [supervisor.updater] Board 'rockpi-4c' not found in version file. No OS updates.
2024-04-06 07:15:15.071 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
2024-04-06 07:29:34.972 INFO (MainThread) [supervisor.hardware.monitor] Detecting remove hardware /dev/ttyUSB0 - /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_a40e24167545ed11a32cc88f0a86e0b4-if00-port0
2024-04-06 07:29:34.984 INFO (MainThread) [supervisor.hardware.monitor] Detecting remove hardware /dev/gpiochip5 - None
```

### Anything in the Host logs that might be useful for us?

```txt
-
```

### Additional information

`# lscpu
Architecture:           aarch64
  CPU op-mode(s):       32-bit, 64-bit
  Byte Order:           Little Endian
CPU(s):                 6
  On-line CPU(s) list:  0-5
Vendor ID:              ARM
  Model name:           Cortex-A53
    Model:              4
    Thread(s) per core: 1
    Core(s) per socket: 4
    Socket(s):          1
    Stepping:           r0p4
    BogoMIPS:           48.00
    Flags:              fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
  Model name:           Cortex-A72
    Model:              2
    Thread(s) per core: 1
    Core(s) per socket: 2
    Socket(s):          1
    Stepping:           r0p2
    BogoMIPS:           48.00
    Flags:              fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
NUMA:
  NUMA node(s):         1
  NUMA node0 CPU(s):    0-5
Vulnerabilities:
  Gather data sampling: Not affected
  Itlb multihit:        Not affected
  L1tf:                 Not affected
  Mds:                  Not affected
  Meltdown:             Not affected
  Mmio stale data:      Not affected
  Retbleed:             Not affected
  Spec rstack overflow: Not affected
  Spec store bypass:    Not affected
  Spectre v1:           Mitigation; __user pointer sanitization
  Spectre v2:           Mitigation; Branch predictor hardening, BHB
  Srbds:                Not affected
  Tsx async abort:      Not affected
`

`# lsmod
Module                  Size  Used by
bnep                   24576  2
nft_chain_nat          12288  27
xfrm_user              45056  1
xfrm_algo              16384  1 xfrm_user
nft_compat             16384  86
nf_tables             249856  1006 nft_compat,nft_chain_nat
nfnetlink              20480  3 nft_compat,nf_tables
cfg80211              438272  0
sch_fq_codel           16384  2
hci_uart              114688  0
cp210x                 24576  1
btqca                  16384  1 hci_uart
btrtl                  24576  1 hci_uart
btbcm                  24576  1 hci_uart
panfrost               65536  0
crct10dif_ce           12288  1
btintel                45056  1 hci_uart
snd_soc_audio_graph_card    12288  0
snd_soc_simple_card    16384  0
rockchipdrm           176128  0
snd_soc_spdif_tx       12288  0
bluetooth             544768  12 btrtl,btqca,btintel,hci_uart,btbcm,bnep
hantro_vpu            266240  0
rockchip_vdec          36864  0
drm_shmem_helper       20480  1 panfrost
ecdh_generic           16384  1 bluetooth
snd_soc_simple_card_utils    28672  2 snd_soc_audio_graph_card,snd_soc_simple_card
analogix_dp            40960  1 rockchipdrm
ecc                    36864  1 ecdh_generic
dw_mipi_dsi            16384  1 rockchipdrm
v4l2_vp9               24576  2 rockchip_vdec,hantro_vpu
gpu_sched              36864  1 panfrost
rfkill                 28672  3 bluetooth,cfg80211
dw_hdmi                49152  1 rockchipdrm
rockchip_rga           24576  0
dwmac_rk               28672  0
v4l2_h264              12288  2 rockchip_vdec,hantro_vpu
videobuf2_dma_sg       16384  1 rockchip_rga
cec                    53248  1 dw_hdmi
stmmac_platform        20480  1 dwmac_rk
snd_soc_rockchip_i2s    20480  2
drm_display_helper    163840  3 rockchipdrm,dw_hdmi,analogix_dp
videobuf2_dma_contig    16384  2 rockchip_vdec,hantro_vpu
v4l2_mem2mem           40960  3 rockchip_vdec,hantro_vpu,rockchip_rga
drm_dma_helper         20480  1 rockchipdrm
videobuf2_memops       12288  2 videobuf2_dma_contig,videobuf2_dma_sg
videobuf2_v4l2         32768  4 rockchip_vdec,hantro_vpu,rockchip_rga,v4l2_mem2mem
drm_kms_helper        208896  6 dw_mipi_dsi,drm_dma_helper,drm_display_helper,rockchipdrm,dw_hdmi,analogix_dp
videobuf2_common       57344  8 rockchip_vdec,videobuf2_dma_contig,videobuf2_v4l2,hantro_vpu,rockchip_rga,videobuf2_dma_sg,v4l2_mem2mem,videobuf2_memops
stmmac                229376  3 stmmac_platform,dwmac_rk
pcs_xpcs               24576  1 stmmac
coresight_cpu_debug    12288  0
coresight              73728  1 coresight_cpu_debug
fuse                  131072  1
drm                   593920  11 gpu_sched,drm_kms_helper,dw_mipi_dsi,drm_dma_helper,drm_shmem_helper,drm_display_helper,rockchipdrm,dw_hdmi,panfrost,analogix_dp
backlight              20480  2 drm_display_helper,drm
`

citruz, 2024-04-07:
Thanks for providing the logs. There should be many more lines in the dmesg log regarding bluetooth initialization. That makes me wonder if there is a problem with the bluetooth chipset in the devicetree. I'm using the 4C+ devicetree from the upstream linux kernel so I would be surprised but it's not impossible.
Can you post the output of `find /proc/device-tree/ | grep blue`?
Just to make sure, this is a Radxa Rock 4C+ board?
Have you tried any of the previous images? 11.5 would be worth testing as there was a kernel upgrade between 11.5 and 12.

mistaoh, 2024-04-08:
Thank you for your work with this.

```
# find /proc/device-tree/ | grep blue
/proc/device-tree/serial@ff180000/bluetooth
/proc/device-tree/serial@ff180000/bluetooth/device-wakeup-gpios
/proc/device-tree/serial@ff180000/bluetooth/pinctrl-names
/proc/device-tree/serial@ff180000/bluetooth/pinctrl-0
/proc/device-tree/serial@ff180000/bluetooth/clock-names
/proc/device-tree/serial@ff180000/bluetooth/clocks
/proc/device-tree/serial@ff180000/bluetooth/shutdown-gpios
/proc/device-tree/serial@ff180000/bluetooth/max-speed
/proc/device-tree/serial@ff180000/bluetooth/compatible
/proc/device-tree/serial@ff180000/bluetooth/vddio-supply
/proc/device-tree/serial@ff180000/bluetooth/vbat-supply
/proc/device-tree/serial@ff180000/bluetooth/host-wakeup-gpios
/proc/device-tree/serial@ff180000/bluetooth/name
#
```
It's [this](https://www.okdo.com/p/okdo-rock-4-model-c-4gb-single-board-computer-starter-kit-with-psu-case-preloaded-linux-os-heat-sink-fan/) board and kit. ROCK 4 Model C+ 4GB single board computer.

I have not tried any of the previous images, and unfortunately i currently don't have physical access to the board for a couple of weeks. But I'm happy to perform any other checks I'm able to do remotely. Otherwise I'll test downgrading at the next opportunity.

citruz, 2024-04-08:
The device tree output looks good and lsmod shows that many of the bluetooth related kernel modules are loaded. `btsdio` is missing but not sure if that is a cause or a symptom. Can you upload the full dmesg log just to make sure there are no other errors?

The next release 12.2 will use the rockchip kernel config from the upstream haos project instead of the one I'm currently maintaining. That might also fix it (for when you have physical access again).

mistaoh, 2024-04-09:
The full dmesg output is attached. I realize when I look through it that there are a lot of errors I dont know what it is.
[dmesg.zip](https://github.com/citruz/haos-rockpi/files/14915911/dmesg.zip)
I also add  journalctl -b 0 --dmesg as I think dmesg buffer fills. 
[journctl.zip](https://github.com/citruz/haos-rockpi/files/14915935/journctl.zip)

Does that mean haos are adding support for this board?

citruz, 2024-04-14:
The dmesg output looks strange indeed. I have never seen this error before
```
es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000016] -6
```

> Does that mean haos are adding support for this board?

No not for the Rock Pi boards. They added some other board that are based on the same rk3399 SoC.

12.2 is out. Let me know if that changes anything once you have physical access again.

https://github.com/citruz/haos-rockpi/releases/tag/12.2%2B20240414

Btw, is Wifi working for you? BT and Wifi are handled by the same chip so I would assume Wifi should be broken as well then.

citruz, 2024-04-14:
What is the output of `cat /proc/device-tree/model`?

mistaoh, 2024-04-15:
Thanks! I'll test it as soon as I'm able.

I have no need for wifi so I never checked. But I don't seem ta have any physical network adapters except `end0` when i check output of `ip addr`

```
# cat /proc/device-tree/model
Radxa ROCK Pi 4C#
```

citruz, 2024-04-15:
> Radxa ROCK Pi 4C

Huh are you sure that you downloaded the correct image `haos_rock-4c-plus`? Because that should return `Radxa ROCK 4C+`

mistaoh, 2024-04-15:
🤪 I feel pretty dumb right now, sorry for wasting your time. I'll test with the correct image.

closed 2024-04-15 by mistaoh

citruz, 2024-04-16:
Haha no worries. There's still a small chance that the wrong devicetree was included with the image but I would be surprised if that is the case.