citruz/haos-rockpi #28: Rock Pi 4A does not boot from emmc  and sd with haos. Also boots from emmc and sd with armbian.

Nikolay-Gamzin, 2024-04-18:
### Describe the issue you are experiencing

When writing an image to an SD card and launching rockpi 4a, a constant reboot occurs, as can be seen from the lights on the keyboard that all blink at once after an equal period of time. Moreover, if you upload armbian to SD or eMMC, everything starts up great

### What operating system image do you use?

rockpi-4a

### What version of Home Assistant Operating System is installed?

tring 9.5,10,11,11.5,latest 12.2

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. install haos_rock-4a_9.5\10\10.5\11\11.5\12.2... on SD card using balena etcher
2. put SD card into rock PI 4A sd slot
power on
...

### Anything in the Supervisor logs that might be useful for us?

```txt
not able to report
```

### Anything in the Host logs that might be useful for us?

```txt
it is not able to start with default setup
```

citruz, 2024-04-26:
What is the output on HDMI?
When you tried with the SD card, did you remove the eMMC module?

Nikolay-Gamzin, 2024-04-26:
Hi.
"What is the output on HDMI?"
I didn't quite understand the question. If you meant what is on the screen, then nothing happens.
"When you tried with the SD card, did you remove the eMMC module?"
Yes, I tried disabling eMMC, but it didn’t work =(
The device restarts cyclically

citruz, 2024-04-27:
Absolutely no output on the screen? Huh, I haven't seen that before.
Sorry, but without serial logs I cannot debug this any further.

Nikolay-Gamzin, 2024-05-18:
@citruz 
U-Boot 2022.07-armbian (Feb 23 2024 - 10:55:08 +0000)

SoC: Rockchip rk3399
Reset cause: RST
Model: Radxa ROCK Pi 4B
DRAM:  3.9 GiB
PMIC:  RK808 
Core:  285 devices, 31 uclasses, devicetree: separate
MMC:   mmc@fe310000: 2, mmc@fe320000: 1, mmc@fe330000: 0
Loading Environment from SPIFlash... SF: Detected XT25F32B with page size 256 Bytes, erase size 4 KiB, total 4 MiB
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Model: Radxa ROCK Pi 4B
Net:   eth0: ethernet@fe300000
Hit any key to stop autoboot:  0 
starting USB...
Bus usb@fe380000: USB EHCI 1.00
Bus usb@fe3c0000: USB EHCI 1.00
Bus usb@fe800000: Register 2000140 NbrPorts 2
Starting the controller
USB XHCI 1.10
Bus usb@fe900000: Register 2000140 NbrPorts 2
Starting the controller
USB XHCI 1.10
scanning bus usb@fe380000 for devices... 1 USB Device(s) found
scanning bus usb@fe3c0000 for devices... 1 USB Device(s) found
scanning bus usb@fe800000 for devices... 1 USB Device(s) found
scanning bus usb@fe900000 for devices... 1 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found

Device 0: unknown device
switch to partitions #0, OK
mmc1 is current device
Scanning mmc 1:1...
Found U-Boot script /boot.scr
3372 bytes read in 5 ms (658.2 KiB/s)
## Executing script at 00500000
switch to partitions #0, OK
mmc1 is current device
loading env...
MMC read: dev # 1, block # 1214464, count 64 ... 64 blocks read: OK
loading environment from haos-config.txt
596 bytes read in 4 ms (145.5 KiB/s)
Unknown command 'fileenv' - try 'help'
Loading standard device tree rk3399-rock-pi-4a.dtb
80822 bytes read in 9 ms (8.6 MiB/s)
329 bytes read in 5 ms (63.5 KiB/s)
Applying kernel provided DT overlay rk3399-disable-wifi-interrupts.dtbo
failed on fdt_overlay_apply(): FDT_ERR_NOTFOUND
Error applying DT overlays, restoring original DT
80822 bytes read in 9 ms (8.6 MiB/s)
Unknown command 'setexpr' - try 'help'
Trying to boot slot A, 3 attempts remaining. Loading kernel ...
Can't set block device
Unknown command 'setexpr' - try 'help'
Trying to boot slot B, 3 attempts remaining. Loading kernel ...
Can't set block device
No valid slot found, resetting tries to 3
storing env...

MMC write: dev # 1, block # 1214464, count 64 ... 64 blocks written: OK
resetting ...

U-Boot TPL 2022.07-armbian (Feb 23 2024 - 10:55:08)
Channel 0: LPDDR4, 50MHz
BW=32 Col=10 Bk=8 CS0 Row=15 CS1 Row=15 CS=2 Die BW=16 Size=2048MB
Channel 1: LPDDR4, 50MHz
BW=32 Col=10 Bk=8 CS0 Row=15 CS1 Row=15 CS=2 Die BW=16 Size=2048MB
256B stride
lpddr4_set_rate: change freq to 400000000 mhz 0, 1
lpddr4_set_rate: change freq to 800000000 mhz 1, 0
Trying to boot from BOOTROM
Returning to boot ROM...

Nikolay-Gamzin, 2024-05-18:
After disconnect eMMC
U-Boot 2022.07-armbian (Feb 23 2024 - 10:55:08 +0000)

SoC: Rockchip rk3399
Reset cause: POR
Model: Radxa ROCK Pi 4B
DRAM:  3.9 GiB
PMIC:  RK808 
Core:  285 devices, 31 uclasses, devicetree: separate
MMC:   mmc@fe310000: 2, mmc@fe320000: 1, mmc@fe330000: 0
Loading Environment from SPIFlash... SF: Detected XT25F32B with page size 256 Bytes, erase size 4 KiB, total 4 MiB
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Model: Radxa ROCK Pi 4B
Net:   eth0: ethernet@fe300000
Hit any key to stop autoboot:  0 
starting USB...
Bus usb@fe380000: USB EHCI 1.00
Bus usb@fe3c0000: USB EHCI 1.00
Bus usb@fe800000: Register 2000140 NbrPorts 2
Starting the controller
USB XHCI 1.10
Bus usb@fe900000: Register 2000140 NbrPorts 2
Starting the controller
USB XHCI 1.10
scanning bus usb@fe380000 for devices... 1 USB Device(s) found
scanning bus usb@fe3c0000 for devices... 1 USB Device(s) found
scanning bus usb@fe800000 for devices... 1 USB Device(s) found
scanning bus usb@fe900000 for devices... 1 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found

Device 0: unknown device
switch to partitions #0, OK
mmc1 is current device
Scanning mmc 1:1...
Found U-Boot script /boot.scr
3372 bytes read in 5 ms (658.2 KiB/s)
## Executing script at 00500000
switch to partitions #0, OK
mmc1 is current device
loading env...
MMC read: dev # 1, block # 1214464, count 64 ... 64 blocks read: OK
loading environment from haos-config.txt
596 bytes read in 4 ms (145.5 KiB/s)
Unknown command 'fileenv' - try 'help'
Loading standard device tree rk3399-rock-pi-4a.dtb
80822 bytes read in 9 ms (8.6 MiB/s)
329 bytes read in 5 ms (63.5 KiB/s)
Applying kernel provided DT overlay rk3399-disable-wifi-interrupts.dtbo
failed on fdt_overlay_apply(): FDT_ERR_NOTFOUND
Error applying DT overlays, restoring original DT
80822 bytes read in 8 ms (9.6 MiB/s)
Unknown command 'setexpr' - try 'help'
Trying to boot slot A, 3 attempts remaining. Loading kernel ...
Can't set block device
Unknown command 'setexpr' - try 'help'
Trying to boot slot B, 3 attempts remaining. Loading kernel ...
Can't set block device
No valid slot found, resetting tries to 3
storing env...

MMC write: dev # 1, block # 1214464, count 64 ... 64 blocks written: OK
resetting ...

U-Boot TPL 2022.07-armbian (Feb 23 2024 - 10:55:08)
Channel 0: LPDDR4, 50MHz
BW=32 Col=10 Bk=8 CS0 Row=15 CS1 Row=15 CS=2 Die BW=16 Size=2048MB
Channel 1: LPDDR4, 50MHz
BW=32 Col=10 Bk=8 CS0 Row=15 CS1 Row=15 CS=2 Die BW=16 Size=2048MB
256B stride
lpddr4_set_rate: change freq to 400000000 mhz 0, 1
lpddr4_set_rate: change freq to 800000000 mhz 1, 0
Trying to boot from BOOTROM
Returning to boot ROM...

U-Boot SPL 2022.07-armbian (Feb 23 2024 - 10:55:08 +0000)
Trying to boot from SPI
NOTICE:  BL31: v1.3(release):845ee93
NOTICE:  BL31: Built : 15:51:11, Jul 22 2020
NOTICE:  BL31: Rockchip release version: v1.1
INFO:    GICv3 with legacy support detected. ARM GICV3 driver initialized in EL3
INFO:    Using opteed sec cpu_context!
INFO:    boot cpu mask: 0
INFO:    If lpddr4 need support multi frequency,
INFO:    please update loader!
INFO:    Current ctl index[0] freq=400MHz
INFO:    Current ctl index[1] freq=800MHz
INFO:    plat_rockchip_pmu_init(1196): pd status 3e
INFO:    BL31: Initializing runtime services
WARNING: No OPTEE provided by BL2 boot loader, Booting device without OPTEE initialization. SMC`s destined for OPTEE will return SMC_UNK
ERROR:   Error initializing runtime service opteed_fast
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x200000
INFO:    SPSR = 0x3c9

citruz, 2024-05-19:
Thanks for the logs, now the problem is pretty clear.
The first line `U-Boot 2022.07-armbian` indicates that your board is still using the armbian bootloader which is not compatible with this image. As you already removed eMMC, it's probably located on the SPI flash. You have to clear the SPI flash so that the board uses the bootloader on the SD card/eMMC.
My board doesn't have that so I'm not 100% sure on the procedure but looking at the [radxa docs](https://wiki.radxa.com/Rockpi4/dev/spi-install) it looks like you can disable the SPI flash at boot by connecting pins 23 and 25. Please try that. If it works you can leave it like that or follow the instructions on the radxa page to completely wipe the SPI flash.

Nikolay-Gamzin, 2024-05-19:
Thanks, all worked well. Can I somehow upload the bootloader to SPI? It takes a long time to load from SDcard =(
P.S.
I tried to clear the SPI and it didn’t work, it only worked by closing contacts 23 and 25 =(
The task can be closed. Thanks again!

citruz, 2024-05-20:
Glad to hear that it works now.
You can try flashing the miniloader image to the SPI flash. That image is only 8MB and contains just the bootloader.
```
sudo rkdeveloptool wl 0 /path/to/miniloader.img
```

closed 2024-05-20 by citruz