citruz/haos-rockpi #33: None of the release images successfully booting on 4C+ V1.41

DuncanDotPY, 2024-05-30:
### Describe the issue you are experiencing

I previously had a very stable haos setup running on a rock 4C+ V1.41 (so thanks!) but after an update to haos, the system would no longer boot anymore. I have a backup I can restore it from, but I can't get any haos image to boot on my 4c+ anymore.

I have tried:

- Multiple Different working SD cards
- All of the older release images
- A whole new rock 4C+ V1.41 board with the above repeated

None of that has worked, implying something upstream perhaps? In any case, I thought it important to flag.
On one (the newly purchased) rock 4c+ the screen stays black, on the other, I get the same behaviour as described in issue 24.

### What operating system image do you use?

rock-4c-plus

### What version of Home Assistant Operating System is installed?

Any

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.Flash an SD card with any of the haos-rockpi releases to a rock 4c+ **V1.41**
2.Attempt to boot

...

### Anything in the Supervisor logs that might be useful for us?

```txt
Can't boot- no access to logs.
```

### Anything in the Host logs that might be useful for us?

```txt
Can't boot- no access to logs.
```

citruz, 2024-06-04:
The fact that previously working images are not working anymore is very surprising. I honestly don't have another idea try since you already checked multiple SD cards and even a new board. I am assuming you don't have an eMMC chip installed?
Let me know when you have the serial logs available.

DuncanDotPY, 2024-06-09:
Yeah no eMMC installed - all SD card based for now. I have a serial adapter ordered and will get around to investigating this as soon as it arrives. Have you any quick and trusty links for setting up logging on the rockpis? If not, I'll work it out of course.

citruz, 2024-06-10:
You can follow the radxa guide: https://wiki.radxa.com/Rockpi4/dev/serial-console
The only thing to note is that this image uses a lower baudrate of 115200 to prevent compatibility issues.

DuncanDotPY, 2024-06-14:
Thanks for the link @citruz. I managed to squeeze some serial logs out of my rock 4c+ boards.
I should add, I've jumped ship onto an RPI 5 but hopefully, this exercise will be useful to someone else.

So I now have two rock 4c+ V1.41 boards. While both have rock 4 c plus v.1.41 printed on the PCB itself, they are visually slightly different (components, labelling etc.). **They also both behave differently.**

**The Newer 4c+**
The new board will only spit out ascii nonsense into the log:
[4cplus_serial_log_New_Board.txt](https://github.com/user-attachments/files/15838001/4cplus_serial_log_New_Board.txt)

**The older 4c+**
The original board that worked once upon a time behaves one of two ways.

1. Without an ethernet cable connecting the board to the router, the board achieves a heartbeat on the blue LED, but still doesn't appear to boot successfully. See 4cplus_serial_log.txt
[4cplus_serial_log.txt](https://github.com/user-attachments/files/15838035/4cplus_serial_log.txt)

2. With an ethernet cable connecting the board to the router, the board doesn't achieve a blue LED heartbeat. See 4cplus_serial_log_with_ethernet.txt
[4cplus_serial_log_with_ethernet.txt](https://github.com/user-attachments/files/15838040/4cplus_serial_log_with_ethernet.txt)

DuncanDotPY, 2024-06-14:
It's the least I can do! I did get nearly a year of stable Home Assistant out of your Images after all :)

> Did you also try the default baudrate of 1500000?

Good call, I'll try that in the morning. It's crashing much earlier so it might give a more fundamental view of the problem.

> I think I didn't ask that yet, do other images like armbian work?

Yes. Armbian, Ubuntu and Debian images all worked on both boards.

citruz, 2024-10-12:
Were you able to solve your problem?

closed 2024-11-09 by citruz