citruz/haos-rockpi #34: Boot failed on Rock Pi 4B+

sebersta, 2024-06-29:
### Describe the issue you are experiencing

It cannot boot from my Rock Pi 4B+ neither by SD card nor NVMe. My device previously booted Armbian on NVMe. I tried booting Armbian on SD card and NVMe, and it all worked out. It just cannot boot HAOS on any media. It did not produce any HDMI signal nor get an IP from Ethernet. The green LED stays on when powered on.

### What operating system image do you use?

rockpi-4b-plus

### What version of Home Assistant Operating System is installed?

12.3

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
not able to report
```

### Anything in the Host logs that might be useful for us?

```txt
not able to report
```

citruz, 2024-07-02:
Did you wipe the eMMC to make sure there is no other bootloader installed there?

citruz, 2024-07-03:
If not, you can boot into any working image and run this to clear the eMMC (use `lsblk` to make sure that mmcblk0 is the eMMC, not the SD card):
```
dd if=/dev/zero of=/dev/mmcblk0 bs=1M count=100
```

sebersta, 2024-07-05:
I wiped the eMMC but it's still not booting. Maybe it has something to do with SPI flash. I followed this guide (Method 1) https://wiki.radxa.com/Rockpi4/install/NVME. Should I remove the SPI flash and try again?

citruz, 2024-07-05:
Yes please try without.
Are you sure you have a 4B+ because that board doesn’t have spi flash https://wiki.radxa.com/Rockpi4/hardware/models

sebersta, 2024-07-05:
It doesn't have a spi flash. I don't even know how it boot armbian from nvme without a bootloader. I use a wire to connect PIN 23 and 25 which block the emmc and I am able to boot HAOS from SD card.

citruz, 2024-07-06:
So booting from SD card works now? That's great.
To boot HAOS from NVMe you need the `miniloader.img` in eMMC and the full image on the NVMe. The setup is described here: https://github.com/citruz/haos-rockpi#option-1-miniloader-in-emmc

sebersta, 2024-07-08:
While Rock Pi 4B+ doesn't have an SPI flash, my retailer soldered it themselves. You have to remove the image from the SPI flash first; at least the official u-boot image doesn't work for HAOS.

closed 2024-07-08 by sebersta