citruz/haos-rockpi #35: No Working Image for ROCK PI 4 PLUS V1.73

Greydesk, 2024-08-14:
### Describe the issue you are experiencing

I have downloaded and installed several images of the latest release. All of them end up without any HDMI output and seem to be in a state of constant reboots with the Ethernet interface coming up for half a second before going down for a second. This happens in time with the Blue LED as well.
I use Linux Mint Mate 22 Wilma as my desktop OS.

### What operating system image do you use?

rockpi-4a-plus

### What version of Home Assistant Operating System is installed?

default

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

The board has the text "ROCK PI 4 PLUS V1.73" silkscreened on it. It has built-in EMMC.

### Steps to reproduce the issue

1. Download an image (tried 4A plus and 4B plus)
2. Burn image onto SD
3. Boot from the SD
...

### Anything in the Supervisor logs that might be useful for us?

```txt
No logs
```

### Anything in the Host logs that might be useful for us?

```txt
No logs
```

citruz, 2024-08-17:
Did you wipe the built-in EMMC to remove any previously installed bootloader? 
https://github.com/citruz/haos-rockpi#installation

Greydesk, 2024-08-20:
You know, I didn't. I assumed that because the device ran LibreELEC fine, it didn't have anything in the EMMC. Took a while to get a working image for something that I could SSH into but I finally cleared it and your image works great!

citruz, 2024-08-20:
No worries. Unfortunately the boatloaders of different images (even if they all use uboot) are not necessarily compatible because it depends on the options that they have been compiled with.
I think there is an alternative way of skipping emmc by grounding a specific pin but it’s not super well documented and i keep forgetting which pin it is. Would probably be a good idea to add that to the readme as it’s a faster workaround than getting another image booted up.

closed 2024-10-07 by citruz