citruz/haos-rockpi #36: HOWTO: Automated update (proposal)

Elarion245, 2024-10-09:
### Describe the issue you are experiencing

Hey all,

I wanted to share my way through which I want to realise as simple as possible updates.

It is a bit hacky, but maybe someone likes it and wants to try... I am not going into all the details, but rather want to explain the high-level concept.

First, create the following file: /homeassistant/haos_update/haos_update_script.sh with the following content (mind unix formating and line-ending) with the following content:
#!/bin/bash

echo -e -n "#!/bin/bash\nrauc install $1\n" > /mnt/data/supervisor/homeassistant/haos_update/perform_update.sh
chmod 777 /mnt/data/supervisor/homeassistant/haos_update/perform_update.sh

Next, activate the Scrape addon to scrape this awesome repo of citruz. 
[image]

The definitions are
RockPi HAOS Available Version:
Ressource: https://github.com/citruz/haos-rockpi/releases
Method: GET
[image]

RockPi HAOS Update Command:
Ressource: https://github.com/citruz/haos-rockpi/releases
Method: GET
[image]

RockPi HAOS URL:
Ressource: https://github.com/citruz/haos-rockpi/releases
Method: GET
[image]

Next, define the following template sensor in configuration.yaml:
  - sensor:
      - name: "RockPi HAOS Current Version"
        unique_id: "RockPi_HAOS_Current"
        icon: "mdi:format-vertical-align-bottom"
        state: >-
          {{ (state_attr('update.home_assistant_operating_system_update', 'installed_version') | regex_findall("(^\d\d\.\d).*") | first) | float }}

This will yield the following entities:
[image]

Also, add the following shell commands to configuration.yaml:
```
  haos_update_script: 'ssh 127.0.0.1 -p22222 -i /config/misc/private_key -o StrictHostKeyChecking=no "/mnt/data/supervisor/homeassistant/haos_update/haos_update_script.sh {{URL}}"'
  haos_perform_update: 'ssh 127.0.0.1 -p22222 -i /config/misc/private_key -o StrictHostKeyChecking=no "/mnt/data/supervisor/homeassistant/haos_update/perform_update.sh"'
 ```

Next, create the following automation:
```
alias: HAOS Update Available
description: ""
triggers:
  - hours: "7"
    trigger: time_pattern
  - hours: "17"
    trigger: time_pattern
conditions:
  - condition: and
    conditions:
      - condition: numeric_state
        entity_id: sensor.rockpi_haos_version
        above: sensor.haos_current_version
actions:
  - data:
      message: >-
        Install via update button on 'Scenarios' or manually via ssh on port 22
        and {{ states('sensor.rockpi_haos_update_command') }}
      title: >-
        HAOS update to version {{ states('sensor.rockpi_haos_version') }}
        available
    enabled: true
    action: notify.persistent_notification
  - data:
      message: >-
        Install via update button on 'Scenarios' or manually via ssh on port 22
        and {{ states('sensor.rockpi_haos_update_command') }}
      title: >-
        HAOS update to version {{ states('sensor.rockpi_haos_version') }}
        available
    action: notify.mobile_app_m2010j19sy
  - action: shell_command.haos_update_script
    data:
      URL: "{{ states('sensor.rockpi_haos_url') }}"
mode: single
```

Finally, you can create the following button as part of your dashboard:
show_name: true
show_icon: true
type: button
tap_action:
  action: call-service
  service: shell_command.haos_perform_update
name: Update RockPi HAOS
icon: mdi:update

When you press this button, nothing visually will happen, apart from the CPU load spiking during the update.
The logical sequence is thus to press the update button whenever the automation makes you aware of a new update of citruz. Then you wait for a minute and restart HAOS completely. You should then find the newest version in the info section.

I hope this helps. Feel free to improve.

### What operating system image do you use?

rock-4c-plus

### What version of Home Assistant Operating System is installed?

13.1

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

N/A

### Steps to reproduce the issue

N/A

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
N/A
```

citruz, 2024-10-12:
Very cool! Thanks for sharing.

Elarion245, 2025-01-30:
Sounds like a simple, yet great improvement.
How do you intend to handle the different boards? As a config of the addon or would it be auto-detected?

In any case: thank you so much for caring about the rock series and Haos!

honsma235, 2025-01-31:
Thank you too for your inspiration!

The board should be auto-detected. With access to the host system the addon can use the [Homeassistant CLI](https://www.home-assistant.io/common-tasks/os/#home-assistant-via-the-command-line).
The command `ha os info` should get you the board.

As a clarification to my previous post:
After updating the addon, the user needs to start it once to trigger the OS update.
Thats because addons aren't restarted automatically after an update, unless they where running allready. But in my opinion it shouldn't always be running in the background. It does nothing, just wastes resources.

Additionaly the script should check and react accordingly on some edge cases:
- current version already installed? -> maybe don't reinstall be default, but allow override with config option
- current addon version older then installed OS? -> probably the same as above
- restore from backup? -> not sure how that works in generall.. if the addon is restores to the same version and state as when it was backed up, then we are good

I just checked and it seems there is no way to select which version of an addon gets installed. Its allways on latest, unless not updated yet. So a downgrade to an older version of the OS would not work out of the box. If needed, it could be implemented through a config option as well.

honsma235, 2025-01-31:
So I got kind of motivated to try that idea out. 
Unless somebody started already..?

honsma235, 2025-02-04:
Well, it wasn't as easy as I thought, but finally I have a working addon: https://github.com/honsma235/haos-rockpi_update-helper
It still needs some polishing and documentation, but feel free to check it out and test/debug:)
Any suggestions welcome.

Elarion245, 2025-02-05:
Great initiative!

For me it ain't working yet. I receive the following error:

```
Download completed successfully.
mount: permission denied (are you root?)
cp: can't stat '/tmp/hassos-overlay/root/.ssh/authorized_keys': No such file or directory
umount: can't unmount /tmp/hassos-overlay: Invalid argument
umount: can't unmount /tmp/tmp: Invalid argument
losetup: /tmp/tmp.img: Operation not permitted
```

Quick question:
Is there any way for the addon to notify us when an update was released? I assume not, as it would need to be running always, right? So the sensors in my op would still make sense...

honsma235, 2025-02-06:
Well, I didn't test it a lot, but never encountered this error. Did you disable the protection mode of the addon?
If the error persists, I will come back to you on how to debug. But I'm busy till beginning of next week. 

Regarding the update: 
Homeassistant regularly checks for new versions of all it components and addons. So as soon as the addon changes to a new version (which in future should be in sync with releases from haos-rockpi), you will get notified. Just like other updates. 
Then you can update the addon and start it once. This will update the OS to the same version as the addon.
The addon just stops after the update and doesn't need to run in the background.

citruz, 2025-02-23:
This is awesome work! I like the creative use of the `bashio::os.config_sync` function to let HAOS handle the update itself.

> Homeassistant regularly checks for new versions of all it components and addons. So as soon as the addon changes to a new version (which in future should be in sync with releases from haos-rockpi), you will get notified. Just like other updates.

This is a great idea to have the add-on updates as a proxy for HAOS updates. Only downside is that you have to manually bump the version every time I do a release. We could check if it's possible to set up cross-repo automation that automatically sets the version on release. Or you give me write access and I would be happy to do it as long as it's only a one-line change.

honsma235, 2025-02-23:
Well, my intention was to transfer the repo to you entirely. That's where it really belongs after all.

The addon is still missing a readme and basic documentation. Maybe a icon/logo too.
And eventually some more testing:)

One more downside I noticed with this aproach: as it is now, there is no simple way to update the addon itself, like to bugfix it.
A good thing is that Home Assistant treats an addon as "new", whenever its version changes somehow. So it work to just shorten the version from the back. It will show up as new, but still find the corresponding OS release. Its just a bit an ugly aproach.

citruz, 2025-02-23:
I think the version compare logic still needs to be fixed. If `OS_VERSION` is greater than `ADDON_VERSION` sort will return exit code 1 which is not what we want. The if condition would need to be flipped but then there's the problem that the exit code is 0 if the versions are equal.

(It just happily downgraded my instance from 14.2 to 14.1)

citruz, 2025-02-23:
```
Current board: rockpi-4b-plus
Current OS version: 14.1.dev20250111
Add-on version: 14.2.dev20250222
[13:26:11] WARNING: 
[13:26:11] WARNING: Current OS version newer or same then add-on version. Please change add-on configuration option 'upgrade_only' to continue.

```

citruz, 2025-02-23:
[Fixed it](https://github.com/citruz/haos-rockpi_update-helper/commit/ea0c8dc9bfa1f6a042db98b22273aa6710573d56) and successfully upgraded my instance from 14.1 -> 14.2 🥳

Btw, I looked into whether we could use the normal haos update infrastructure for this fork as well but it would require me to publish patched versions of (at least) the supervisor docker image with a different update url. That means maintaining another forked repo and mirroring releases for it. That's a lot of additional work so I really appreciate this creative solution.