citruz/haos-rockpi #37: Copying image to spl.0

justnx, 2024-10-11:
### Describe the issue you are experiencing

Use `ha` to access the Home Assistant CLI.
# rauc install https://github.com/citruz/haos-rockpi/releases/download/13.1%2B20241005/haos_rockpi-4b-13.1.dev20241005.raucb
installing
  0% Installing
  0% Determining slot states
 10% Determining slot states done.
 10% Checking bundle
 10% Verifying signature
 20% Verifying signature done.
 20% Checking bundle done.
 20% Checking manifest contents
 30% Checking manifest contents done.
 30% Determining target install group
 40% Determining target install group done.
 40% Updating slots
 40% Checking slot boot.0
 41% Checking slot boot.0 done.
 41% Copying image to boot.0
 55% Copying image to boot.0 done.
 55% Checking slot kernel.1
 56% Checking slot kernel.1 done.
 56% Copying image to kernel.1
 57% Copying image to kernel.1
 58% Copying image to kernel.1
 59% Copying image to kernel.1
 60% Copying image to kernel.1
 61% Copying image to kernel.1
 62% Copying image to kernel.1
 63% Copying image to kernel.1
 64% Copying image to kernel.1
 65% Copying image to kernel.1
 66% Copying image to kernel.1
 67% Copying image to kernel.1
 68% Copying image to kernel.1
 69% Copying image to kernel.1
 70% Copying image to kernel.1 done.
 70% Checking slot rootfs.1
 71% Checking slot rootfs.1 done.
 71% Copying image to rootfs.1
 72% Copying image to rootfs.1
 73% Copying image to rootfs.1
 74% Copying image to rootfs.1
 75% Copying image to rootfs.1
 76% Copying image to rootfs.1
 77% Copying image to rootfs.1
 78% Copying image to rootfs.1
 79% Copying image to rootfs.1
 80% Copying image to rootfs.1
 81% Copying image to rootfs.1
 82% Copying image to rootfs.1
 83% Copying image to rootfs.1
 84% Copying image to rootfs.1
 84% Copying image to rootfs.1 done.
 84% Checking slot spl.0
 86% Checking slot spl.0 done.
 86% Copying image to spl.0
 99% Copying image to spl.0 failed.
 99% Updating slots failed.
100% Installing failed.
LastError: Installation error: Failed updating slot spl.0: failed to run slot hook: Child process exited with code 1
idle
Installing `https://github.com/citruz/haos-rockpi/releases/download/13.1%2B20241005/haos_rockpi-4b-13.1.dev20241005.raucb` failed
# 

### What operating system image do you use?

rockpi-4b

### What version of Home Assistant Operating System is installed?

core-2024.10.1

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

rockpi 4b OS installed on nvme

### Steps to reproduce the issue

run rauc install on rockpi 4b

### Anything in the Supervisor logs that might be useful for us?

```txt
no
```

### Anything in the Host logs that might be useful for us?

```txt
no
```

### System information

Home Assistant Supervisor
Home Assistant OS 12.3.dev20240509
Update-Channel | stable
Supervisor-Version | supervisor-2024.10.0
Agent-Version | 1.6.0
Docker-Version | 25.0.5

citruz, 2024-10-12:
Can you post the output of `journalctl -u rauc.service` after running `rauc install`? there should be an error.
Since you're using an ssd, where do you have the miniloader installed?

justnx, 2024-10-12:
On mmc

There you go:
Oct 12 07:38:24 homeassistant rauc[513]: Reading bundle: /mnt/data/tmp/rauc-GFZMV2/download.raucb
Oct 12 07:38:24 homeassistant rauc[513]: Detected CRL but CRL checking is disabled!
Oct 12 07:38:24 homeassistant rauc[513]: Verifying bundle signature... 
Oct 12 07:38:24 homeassistant rauc[513]: Verified inline signature by 'CN = hassos-rockpi rauc OTA'
Oct 12 07:38:24 homeassistant rauc[513]: Mounting bundle '/mnt/data/tmp/rauc-GFZMV2/download.raucb' to '/run/rauc/bundle'
Oct 12 07:38:24 homeassistant rauc[513]: Configured loop device '/dev/loop0' for 172838912 bytes
Oct 12 07:38:24 homeassistant rauc[513]: Configured dm-verity device '/dev/dm-0'
Oct 12 07:38:24 homeassistant rauc[513]: Running bundle hook install-check
Oct 12 07:38:24 homeassistant rauc[513]: Checking image type for slot type: raw
Oct 12 07:38:24 homeassistant rauc[513]: Image detected as type: *.img
Oct 12 07:38:24 homeassistant rauc[513]: Checking image type for slot type: raw
Oct 12 07:38:24 homeassistant rauc[513]: Image detected as type: *.img
Oct 12 07:38:24 homeassistant rauc[513]: Marking target slot kernel.1 as non-bootable...
Oct 12 07:38:24 homeassistant rauc[513]: Marked slot kernel.1 as bad
Oct 12 07:38:24 homeassistant rauc[513]: installing https://github.com/citruz/haos-rockpi/releases/download/13.1%2B20241005/haos_rockpi-4b-13.1.dev20241005.raucb: Updating slots...
Oct 12 07:38:24 homeassistant rauc[513]: installing https://github.com/citruz/haos-rockpi/releases/download/13.1%2B20241005/haos_rockpi-4b-13.1.dev20241005.raucb: Checking slot boot.0
Oct 12 07:38:24 homeassistant rauc[513]: Updating slot boot.0
Oct 12 07:38:24 homeassistant rauc[513]: Updating /dev/disk/by-partlabel/hassos-boot with 'install' slot hook
Oct 12 07:38:24 homeassistant rauc[513]: installing https://github.com/citruz/haos-rockpi/releases/download/13.1%2B20241005/haos_rockpi-4b-13.1.dev20241005.raucb: Updating slot boot.0
Oct 12 07:38:24 homeassistant rauc[513]: Running slot hook 'slot-install' for boot.0
Oct 12 07:38:24 homeassistant rauc[1956]: cp: can't stat '/mnt/boot/EFI/BOOT/grubenv': No such file or directory
Oct 12 07:38:24 homeassistant rauc[1962]: cp: can't stat '/tmp/boot-tmp/grubenv': No such file or directory
Oct 12 07:38:24 homeassistant rauc[513]: Updating slot boot.0 status
Oct 12 07:38:24 homeassistant rauc[513]: installing https://github.com/citruz/haos-rockpi/releases/download/13.1%2B20241005/haos_rockpi-4b-13.1.dev20241005.raucb: Updating slot boot.0 done
Oct 12 07:38:24 homeassistant rauc[513]: installing https://github.com/citruz/haos-rockpi/releases/download/13.1%2B20241005/haos_rockpi-4b-13.1.dev20241005.raucb: Checking slot kernel.1
Oct 12 07:38:24 homeassistant rauc[513]: Updating slot kernel.1
Oct 12 07:38:24 homeassistant rauc[513]: installing https://github.com/citruz/haos-rockpi/releases/download/13.1%2B20241005/haos_rockpi-4b-13.1.dev20241005.raucb: Updating slot kernel.1
Oct 12 07:38:24 homeassistant rauc[513]: Updating /dev/disk/by-partlabel/hassos-kernel1 with /run/rauc/bundle/kernel.img
Oct 12 07:38:24 homeassistant rauc[513]: opening slot device /dev/disk/by-partlabel/hassos-kernel1
Oct 12 07:38:24 homeassistant rauc[513]: writing data to device /dev/disk/by-partlabel/hassos-kernel1
Oct 12 07:38:25 homeassistant rauc[513]: Updating slot kernel.1 status
Oct 12 07:38:25 homeassistant rauc[513]: installing https://github.com/citruz/haos-rockpi/releases/download/13.1%2B20241005/haos_rockpi-4b-13.1.dev20241005.raucb: Updating slot kernel.1 done
Oct 12 07:38:25 homeassistant rauc[513]: installing https://github.com/citruz/haos-rockpi/releases/download/13.1%2B20241005/haos_rockpi-4b-13.1.dev20241005.raucb: Checking slot rootfs.1
Oct 12 07:38:25 homeassistant rauc[513]: Updating slot rootfs.1
Oct 12 07:38:25 homeassistant rauc[513]: Updating /dev/disk/by-partlabel/hassos-system1 with /run/rauc/bundle/rootfs.img
Oct 12 07:38:25 homeassistant rauc[513]: opening slot device /dev/disk/by-partlabel/hassos-system1
Oct 12 07:38:25 homeassistant rauc[513]: installing https://github.com/citruz/haos-rockpi/releases/download/13.1%2B20241005/haos_rockpi-4b-13.1.dev20241005.raucb: Updating slot rootfs.1
Oct 12 07:38:25 homeassistant rauc[513]: writing data to device /dev/disk/by-partlabel/hassos-system1
Oct 12 07:38:27 homeassistant rauc[513]: rauc mark: already processing a different method
Oct 12 07:38:29 homeassistant rauc[513]: Updating slot rootfs.1 status
Oct 12 07:38:29 homeassistant rauc[513]: installing https://github.com/citruz/haos-rockpi/releases/download/13.1%2B20241005/haos_rockpi-4b-13.1.dev20241005.raucb: Updating slot rootfs.1 done
Oct 12 07:38:29 homeassistant rauc[513]: installing https://github.com/citruz/haos-rockpi/releases/download/13.1%2B20241005/haos_rockpi-4b-13.1.dev20241005.raucb: Checking slot spl.0
Oct 12 07:38:29 homeassistant rauc[513]: Updating slot spl.0
Oct 12 07:38:29 homeassistant rauc[513]: Updating /dev/disk/by-partlabel/hassos-boot with 'install' slot hook
Oct 12 07:38:29 homeassistant rauc[513]: installing https://github.com/citruz/haos-rockpi/releases/download/13.1%2B20241005/haos_rockpi-4b-13.1.dev20241005.raucb: Updating slot spl.0
Oct 12 07:38:29 homeassistant rauc[513]: Running slot hook 'slot-install' for spl.0
Oct 12 07:38:30 homeassistant rauc[2456]: 16320+0 records in
Oct 12 07:38:30 homeassistant rauc[2456]: 16320+0 records out
Oct 12 07:38:30 homeassistant rauc[2456]: 8355840 bytes (8.0MB) copied, 0.706087 seconds, 11.3MB/s
[... cut ...]
Oct 12 07:38:30 homeassistant rauc[513]: installing https://github.com/citruz/haos-rockpi/releases/download/13.1%2B20241005/haos_rockpi-4b-13.1.dev20241005.raucb: Installation error: Failed updating slot spl.0: failed to run slot hook: Child process exited with code 1
Oct 12 07:38:30 homeassistant rauc[513]: installing https://github.com/citruz/haos-rockpi/releases/download/13.1%2B20241005/haos_rockpi-4b-13.1.dev20241005.raucb: finished
Oct 12 07:38:30 homeassistant rauc[513]: installing `https://github.com/citruz/haos-rockpi/releases/download/13.1%2B20241005/haos_rockpi-4b-13.1.dev20241005.raucb` failed: 1

justnx, 2024-10-13:
doing a clean install by writing new latest image to nvme also seems not to help here.

citruz, 2024-10-13:
I built a safeguard into the updater so that it aborts when the block device/partition layout is unexpected to avoid overwriting the wrong device.
It expects `mmcblk0` but in your case it seems to  be `mmcblk1` (your mmc is 32GB?). And there’s another device `mtdblock0` which I haven’t seen before.

justnx, 2024-10-13:
mtdblock0 is the spi flash drive from the rockpi itself, see: https://wiki.radxa.com/Rockpi4/dev/spi-install

regarding "mmcblk0". interesting. how can i overwrite that behavior?
it worked just fine before until a specific release.

Disk /dev/mtdblock0: 16 MiB, 16777216 bytes, 32768 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes

Disk /dev/mmcblk1: 29.72 GiB, 31914983424 bytes, 62333952 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes

citruz, 2024-10-19:
> it worked just fine before until a specific release.

That's interesting. I haven't changed that code since I added it. I wonder if something in the kernel configuration changed so that the mtdblock0 device appeared and mmcblk0 changed to mmcblk1.

The behavior can't be changed currently but for 13.2 I'll change it to also allow mmcblk1 as a target.

justnx, 2024-10-22:
that would be awesome if you could consider that in your new release.

yeah its totally possible that newer kernel brings changes in device mapping. 
thats also the reason why to use uuid in fstab for mount points nowdays instead dev naming.

citruz, 2024-11-09:
Just published 13.2 which adds the fallback to mmcblk1. Please let me know if that fixes your issue.

justnx, 2024-11-11:
works perfect again now. thanks for the update

closed 2024-11-11 by justnx