citruz/haos-rockpi #40: Merge into Haos?

Elarion245, 2025-01-03:
### Describe the issue you are experiencing

Hey citruz,

I know that some time ago you weren't a fan of integrating your work into the official Haos, but don't you think that now this may have changed? your builds are super stable and they bring so much value.
Nevertheless each release is work on your end and without you Haos on Rock is basically dead...
I would like to reopen the merging idea, if you are open for it 😉

### What operating system image do you use?

rock-4c-plus

### What version of Home Assistant Operating System is installed?

13.2

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

N/A

### Steps to reproduce the issue

N/A

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
N/A
```

### System information

N/A

### Additional information

N/A

citruz, 2025-01-13:
Hey,
It's not really that I don't want it but I'm not sure how the haos folks feel about supporting a family of 7 new devices 😅
Having that said, I never "officially" inquired about what would be necessary to merge it upstream so I can definitely do that.

Elarion245, 2025-01-13:
I'd say 'give it a try'. Work like yours is the spirit of HA. If they are afraid, then they can do one after the other 😉
By the way, many thanks for 14.1!